bulletphysics/bullet3 #1768: FontData.h:265:1: narrowing conversion of '-96' from 'int' to 'char' inside { }

yurivict, 2018-06-18:
Build fails on FreeBSD on the powerpc64 architecture. The port got flagged with this error message.
I'm not sure which compiler is this, I didn't seethe error myself.

CptMonkeyFist, 2018-08-10:
I just ran into this error when compiling for ARMv6 (Raspberry Pi Zero W) using Gnu compilers version 6.3.0.
1. cmake -G"Unix Makefiles" -DCMAKE_BUILD_TYPE=Release
2. make

The fault appears to lie in building the Gwen third party library in my case.

```
In file included from /home/pi/bullet3/examples/ThirdPartyLibs/Gwen/Renderers/Op                                                                                                                                                      enGL_DebugFont.cpp:37:0:
/home/pi/bullet3/examples/ThirdPartyLibs/Gwen/Renderers/FontData.h:265:1: error:                                                                                                                                                       narrowing conversion of ‘-96’ from ‘int’ to ‘char’ inside { } [-Wnarrowing]
 };
 ^
/home/pi/bullet3/examples/ThirdPartyLibs/Gwen/Renderers/FontData.h:265:1: error:                                                                                                                                                       narrowing conversion of ‘-48’ from ‘int’ to ‘char’ inside { } [-Wnarrowing]
/home/pi/bullet3/examples/ThirdPartyLibs/Gwen/Renderers/FontData.h:265:1: error:                                                                                                                                                       narrowing conversion of ‘-64’ from ‘int’ to ‘char’ inside { } [-Wnarrowing]
/home/pi/bullet3/examples/ThirdPartyLibs/Gwen/Renderers/FontData.h:265:1: error:                                                                                                                                                       narrowing conversion of ‘-96’ from ‘int’ to ‘char’ inside { } [-Wnarrowing]
/home/pi/bullet3/examples/ThirdPartyLibs/Gwen/Renderers/FontData.h:265:1: error:                                                                                                                                                       narrowing conversion of ‘-96’ from ‘int’ to ‘char’ inside { } [-Wnarrowing]
/home/pi/bullet3/examples/ThirdPartyLibs/Gwen/Renderers/FontData.h:265:1: error:                                                                                                                                                       narrowing conversion of ‘-48’ from ‘int’ to ‘char’ inside { } [-Wnarrowing]
/home/pi/bullet3/examples/ThirdPartyLibs/Gwen/Renderers/FontData.h:265:1: error:                                                                                                                                                       narrowing conversion of ‘-64’ from ‘int’ to ‘char’ inside { } [-Wnarrowing]
/home/pi/bullet3/examples/ThirdPartyLibs/Gwen/Renderers/FontData.h:265:1: error:                                                                                                                                                       narrowing conversion of ‘-16’ from ‘int’ to ‘char’ inside { } [-Wnarrowing]
/home/pi/bullet3/examples/ThirdPartyLibs/Gwen/Renderers/FontData.h:265:1: error:                                                                                                                                                       narrowing conversion of ‘-128’ from ‘int’ to ‘char’ inside { } [-Wnarrowing]
/home/pi/bullet3/examples/ThirdPartyLibs/Gwen/Renderers/FontData.h:265:1: error:                                                                                                                                                       narrowing conversion of ‘-80’ from ‘int’ to ‘char’ inside { } [-Wnarrowing]
/home/pi/bullet3/examples/ThirdPartyLibs/Gwen/Renderers/FontData.h:265:1: error:                                                                                                                                                       narrowing conversion of ‘-16’ from ‘int’ to ‘char’ inside { } [-Wnarrowing]
/home/pi/bullet3/examples/ThirdPartyLibs/Gwen/Renderers/FontData.h:265:1: error:                                                                                                                                                       narrowing conversion of ‘-64’ from ‘int’ to ‘char’ inside { } [-Wnarrowing]
/home/pi/bullet3/examples/ThirdPartyLibs/Gwen/Renderers/FontData.h:265:1: error:                                                                                                                                                       narrowing conversion of ‘-64’ from ‘int’ to ‘char’ inside { } [-Wnarrowing]
/home/pi/bullet3/examples/ThirdPartyLibs/Gwen/Renderers/FontData.h:265:1: error:                                                                                                                                                       narrowing conversion of ‘-32’ from ‘int’ to ‘char’ inside { } [-Wnarrowing]
/home/pi/bullet3/examples/ThirdPartyLibs/Gwen/Renderers/FontData.h:265:1: error:                                                                                                                                                       narrowing conversion of ‘-16’ from ‘int’ to ‘char’ inside { } [-Wnarrowing]
/home/pi/bullet3/examples/ThirdPartyLibs/Gwen/Renderers/FontData.h:265:1: error:                                                                                                                                                       narrowing conversion of ‘-48’ from ‘int’ to ‘char’ inside { } [-Wnarrowing]
[... cut ...]
make[2]: *** [examples/ThirdPartyLibs/Gwen/CMakeFiles/gwen.dir/Renderers/OpenGL_                                                                                                                                                      DebugFont.o] Error 1
CMakeFiles/Makefile2:830: recipe for target 'examples/ThirdPartyLibs/Gwen/CMakeF                                                                                                                                                      iles/gwen.dir/all' failed
make[1]: *** [examples/ThirdPartyLibs/Gwen/CMakeFiles/gwen.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
```

yurivict, 2018-08-10:
@CptMonkeyFist Disable examples with the cmake option.

CptMonkeyFist, 2018-08-11:
That solved this problem for me and while I understand why demos may not work on an ARM device, the author of the issue would probably prefer to have them. I am having other issues, but will post them in a separate Github issue at a later time if I cannot find an answer in previous issues. Thanks yurivict.

closed 2021-04-26 by erwincoumans