bulletphysics/bullet3 #1772: MJCF file; joints not responding properly

Anmol6, 2018-06-22:
Hi, [here's](https://gist.github.com/Anmol6/5520170dfa35a958ca69fbf854044203) a mjcf file that I'm loading in pybullet with the following code:
```
import pybullet as p;
import pybullet_data; 
p.connect(p.GUI);
p.setAdditionalSearchPath(pybullet_data.getDataPath()); 
planeID=p.loadURDF("plane.urdf");  
bid=p.loadMJCF('2dbiped.xml');
p.setGravity(0,0 ,-9.8); 
p.setRealTimeSimulation(True)
```
I noticed the following problems:
1) the torso always wants to stay upright
2) when the leg is lifted, it slowly does move down but not nearly as fast as it should

Any insight into this would be appreciated; thanks!

erwincoumans, 2018-06-22:
No idea, we focus primarily on URDF (we just added a rudimentary MJCF loader for some OpenAI Gym models). Attached is a URDF you could try to use as a starting point instead.
[biped_pybullet_urdf_debug (1).zip](https://github.com/bulletphysics/bullet3/files/2129294/biped_pybullet_urdf_debug.1.zip)

closed 2018-06-22 by erwincoumans

Anmol6, 2018-06-22:
@erwincoumans Thanks for that! However, I am not able to add a 2d movement constraint using planar joints to the body (I also tried adding just one planar constraint). I get a segmentation fault when I load [this](https://gist.github.com/Anmol6/7370676bda259a0f11968420b1ac5bb0) file. Any ideas on how to add this constraint?

Anmol6, 2018-06-22:
It might be better if I open a new issue, actually.

erwincoumans, 2018-06-23:
We start with a fixed base, then add prismatic and revolute joints. This is how we convert MuJoCo 2D models too (2D hopper and Half Cheetah etc). We don't expose/support planar constraints and spherical joints (none of our robots have them).

jkoendev, 2019-08-14:
@erwincoumans i would be very interested in having URDF files of 2D hopper and half cheetah! Are these available somewhere online, or could you provide them to me? Best, Jonas

erwincoumans, 2019-08-14:
2d hopper, 2d half cheetah, humanoid, ant etc are all part of PyBullet pybullet_envs and pybullet_data including their Gym environments and pre-trained policies/neural network weights.

MJCF files are here:
https://github.com/bulletphysics/bullet3/tree/master/examples/pybullet/gym/pybullet_data/mjcf

~~~~~~~~~~~~~~
pip3 install pybullet
python3 -m pybullet_envs.examples.enjoy_TF_HopperBulletEnv_v0_2017ma
python3 -m pybullet_envs.examples.enjoy_TF_AntBulletEnv_v0_2017may
python3 -m pybullet_envs.examples.enjoy_TF_HumanoidBulletEnv_v0_2017may
python3 -m pybullet_envs.examples.enjoy_TF_HumanoidFlagrunHarderBulletEnv_v1_2017jul
~~~~~~~~~~~~~~

[2d_hopper_cheetah.zip](https://github.com/bulletphysics/bullet3/files/3501893/2d_hopper_cheetah.zip)

You can use the mjcf2urdf.py utility

https://github.com/bulletphysics/bullet3/blob/master/examples/pybullet/gym/pybullet_utils/examples/mjcf2urdf.py

jkoendev, 2019-08-14:
this is great, thanks a lot!

erwincoumans, 2019-08-14:
Don't expect too much, the mjcf2urdf is a quick rudimentary hack, only covering a small subset of mjcf that was used to get the gym environments working.