bulletphysics/bullet3 #1773: URDF File; Adding planar constraint

Anmol6, 2018-06-22:
Hi, I am not able to add a planar movement constraint using planar joints for a biped (I also tried adding just one planar constraint). I get a segmentation fault when I load [this](https://gist.github.com/Anmol6/7370676bda259a0f11968420b1ac5bb0) file. Any ideas on how to add this constraint?

erwincoumans, 2018-06-23:
Here is an example of the biped in 2d, using 2 prismatic and 1 revolute joint. We don't expose planar nor spherical joints at the moment (no short-term resources to add them, since non of our robots need them -> lower priority)
[test2d.zip](https://github.com/bulletphysics/bullet3/files/2129495/test2d.zip)

Anmol6, 2018-06-23:
Super - thank you! I was trying to add two prismatic joints to the same body earlier, instead of following a tree-like structure.

closed 2018-06-23 by Anmol6

Anmol6, 2018-06-23:
Actually I noticed another thing; you set the mass of the 'world' object to 0. I tried setting it to something really small and it totally makes the 2d constraint go away... Why is that?

erwincoumans, 2018-06-23:
If the world has any non-zero mass, the world will keep on falling down :-)

Anmol6, 2018-06-23:
Right, and zero-mass also makes the link immune to force/torque applied by an attached link?

erwincoumans, 2018-06-23:
Yes, a zero-mass base is considered 'fixed' (infinite mass).