bulletphysics/bullet3 #1777: pybullet -  loadURDF changes AABB of next created multibody

rsthomp, 2018-06-26:
When I load a URDF file, then create an collisionShape, the reported bounding box of the collisionShape is the bounding box of the URDF object. It looks like this happens once per URDF (i.e. if I load two URDFs, then create two collisionShapes, both of the shape's bounding boxes are incorrect)

For example, running this: 
```
r2d2 = p.loadURDF("r2d2.urdf", basePosition=[-1,0,1])

r2d3 = p.loadURDF("r2d2.urdf", basePosition=[0,0,1])
print(p.getAABB(r2d2))
print(p.getAABB(r2d3))

print("")
#Obstacle 1 - same bounding box as r2d2
obstId = p.createCollisionShape(p.GEOM_BOX, halfExtents = [.1, .1, .1], )
p.createMultiBody(baseCollisionShapeIndex=obstId, basePosition=[-1, -1, .1])

#Obstacle 2 - same bounding box as r2d3
obstId2 = p.createCollisionShape(p.GEOM_BOX, halfExtents = [.1, .1, .1], )
p.createMultiBody(baseCollisionShapeIndex=obstId2, basePosition=[-1, -1, .1])

#Obstacle 3 - correct bounding box
obstId3 = p.createCollisionShape(p.GEOM_BOX, halfExtents = [.1, .1, .1], )
p.createMultiBody(baseCollisionShapeIndex=obstId3, basePosition=[-1, -1, .1])

print(p.getAABB(obstId))
print(p.getAABB(obstId2))
print(p.getAABB(obstId3))

while 1:
	p.stepSimulation()
```

Produces this output:
```
((-1.172, -0.17200000000000001, 0.698), (-0.828, 0.17200000000000001, 1.302))
((-0.17200000000000001, -0.17200000000000001, 0.698), (0.17200000000000001, 0.17200000000000001, 1.302))

((-1.172, -0.17200000000000001, 0.698), (-0.828, 0.17200000000000001, 1.302))
((-0.17200000000000001, -0.17200000000000001, 0.698), (0.17200000000000001, 0.17200000000000001, 1.302))
((-1.1, -1.1, 0.0), (-0.9, -0.9, 0.2))
```

When it should produce:
```
((-1.172, -0.17200000000000001, 0.698), (-0.828, 0.17200000000000001, 1.302))
((-0.17200000000000001, -0.17200000000000001, 0.698), (0.17200000000000001, 0.17200000000000001, 1.302))

((-1.1, -1.1, 0.0), (-0.9, -0.9, 0.2))
((-1.1, -1.1, 0.0), (-0.9, -0.9, 0.2))
((-1.1, -1.1, 0.0), (-0.9, -0.9, 0.2))
```
Despite all the shapes being displayed correctly.

erwincoumans, 2018-06-26:
p.getAABB takes the object unique id, not the shape id.

closed 2018-06-26 by erwincoumans

rsthomp, 2018-06-26:
Ok, thank you!

erwincoumans, 2018-06-26:
Sure, you are welcome, hopy you enjoy PyBullet a bit :)