bulletphysics/bullet3 #1779: Pybullet - request feature: ellipsoid geometric shape

bdelhaisse, 2018-07-01:
Hi,

Would it be possible for pybullet to provide access to an ellipsoid visual and collision shape? I saw that it can be done in C++ using `btMultiSphereShape` (on https://pybullet.org/Bullet/phpBB3/viewtopic.php?t=808), however through pybullet we don't have access to that shape. Was there a particular reason for this?

erwincoumans, 2018-09-05:
We haven't added it since we didn't need it yet and ellipsoids are not exposed in the URDF format, which is our main format in robotics. Until it is exposed, you can easily create a tesselated ellipsoid as convex mesh.

Can you tell what you are using it for?

bdelhaisse, 2018-09-06:
Hi, I use it to draw manipulability ellipsoids. For now, I created a 3D mesh of an ellipsoid which is loaded into the simulator. However, my biggest problem right now is that when the robot is moving, the manipulability ellipsoid (which depends on the Jacobian) is changing as well (its position, orientation and scaling), but pybullet doesn't allow you to rescale a 3D mesh in an online manner. So, I am currently removing and reloading the ellipsoids for each end effector as the robot is moving, but this is not smooth, and is certainly not the proper way to do it. This is related to issue #1816.

erwincoumans, 2018-12-22:
We have capsule support in PyBullet URDF, that's all we got.
See Bullet/data/capsule.urdf

closed 2018-12-22 by erwincoumans

TFLQW, 2022-03-09:
> Hi!
> 
> I am not sure if you have figured this out, but this is, in fact, a relevant functionality that is missing in Pybullet (**I am also in need of it**). It would be nice to add this functionality to Pybullet or to know if there is a way to do it.

Hi, did you solve it, I also have that problem!

SirNate0, 2024-01-15:
@erwincoumans is it still possible to use a multi sphere shape as an ellipsoid, as was suggested many years ago?

https://pybullet.org/Bullet/phpBB3/viewtopic.php?p=2822&sid=fb649a2193c4c61a217adbd1ea584ef9#p2822