bulletphysics/bullet3 #1783: Errors when linking with Cmake

Flibio, 2018-07-07:
Hello, I am trying to integrate Bullet into my cmake project, but every time I try to, I receive errors when building the project. I am on Windows 10 and use the Clion IDE.

**Errors thrown:**

```
CMakeFiles\Artemis.dir/objects.a(main.cpp.obj): In function `ZN18btAlignedAllocatorIiLj16EE10deallocateEPi':
C:/bullet/master/src/LinearMath/btAlignedAllocator.h:96: undefined reference to `btAlignedFreeInternal(void*)'
CMakeFiles\Artemis.dir/objects.a(main.cpp.obj): In function `ZN18btAlignedAllocatorI9btHashIntLj16EE10deallocateEPS0_':
C:/bullet/master/src/LinearMath/btAlignedAllocator.h:96: undefined reference to `btAlignedFreeInternal(void*)'
CMakeFiles\Artemis.dir/objects.a(main.cpp.obj): In function `ZN18btAlignedAllocatorI14btTriangleInfoLj16EE10deallocateEPS0_':
C:/bullet/master/src/LinearMath/btAlignedAllocator.h:96: undefined reference to `btAlignedFreeInternal(void*)'
```

**Steps taken to compile Bullet:**

1. Used Cmake-GUI to generate build files with the `MinGW Makefiles` generator.
1. Ran the command `make` to successfully build the project

**My Cmake File:**

```cmake
cmake_minimum_required(VERSION 3.8)
project(Artemis)

set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -std=c++14 -static -g")
set(CMAKE_CXX_FLAGS_DEBUG "${CMAKE_CXX_FLAGS_DEBUG} -O0 -g")
set(CMAKE_C_FLAGS_DEBUG "${CMAKE_C_FLAGS_DEBUG} -O0 -g")

set(CMAKE_MODULE_PATH ${CMAKE_CURRENT_SOURCE_DIR}/cmake_modules/)

set(BULLET_ROOT "C:/bullet/master/build")
set(BULLET_INCLUDE_DIR "C:/bullet/master/src")

find_package(Bullet REQUIRED)
include_directories(${BULLET_INCLUDE_DIR})

set(SOURCE_FILES src/main.cpp)

add_executable(Artemis ${SOURCE_FILES})

target_link_libraries(Artemis ${BULLET_LIBRARIES})
```

termi3, 2018-07-08:
Try replacing "BULLET" with "BULLET_PHYSICS",
check the project name in the cmakelists of the bullet lib there is PROJECT(BULLET_PHYSICS)
If this does not help, check the "Linker Options" in your IDE (I do not have experience with CLion) and have a look what links there, here definitivelly is missing the proper LINK to the bullet libs

termi3, 2018-07-08:
There should be Linker Options or some other place, where you define what to link, the question is how these are sent to the commman line. I have from time to time similar problems with Visual Studio, but all my projects are based on CMake. CMake generate VS projekt file and then you can see (in Visual Studio - Project Properties -> linker) what should be linked there;
In the older time I had as well problems compiling Bullets with minGW, so this was the reason to switch to VS;
I know what the very last line is doing but you should check what ${BULLET_LIBRARIES} has and if the compiled libraries are the proper (debug/release 32/64 bit).
You can check wich command line parameters are send to  the linker  and if there are listed the bullet libarires.
OR
you could include Bullets in your project three and compile it as library... Takes more time but solves several problems (if you have to swich often between different options dll/lib 32/64 etc.)

Flibio, 2018-07-08:
I much prefer CLion of Visual Studio, so for now I have just included the Bullet files I need in my project. It isn't a very good solution though and I would like to resolve this problem.

I've checked `${BULLET_LIBRARIES}` and it contains all the of the `.a` files it should. I'm not generating `.dll` or `.lib` files, instead `.a` files. The actual compilation of Bullet works just fine and the `.a` files are generated. The problem seems to be with getting it linked to my project which is building with pure Cmake, no GUI for linker options.

termi3, 2018-07-08:
So, have probably to add "install" 
https://cmake.org/cmake/help/v3.0/command/install.html
after compiling the bullet, so that you compiler find these, it can be issues of the path.

Flibio, 2018-07-08:
I've checked the paths, they are all good.

erwincoumans, 2018-12-22:
Can't help with this. Linking Bullet with cmake for all our platforms works just fine.

closed 2018-12-22 by erwincoumans

eadgyo, 2023-09-09:
Same error here. Resolved for me by including libLinearMath.a, which is not named as a bullet lib. As a cmake argument in target_link_libraries, we should add -lLinearMath.

JoshikaNetha, 2025-05-02:
Hi,
I'm trying to integrate Bullet Physics into my Qt/QML application. I'm new to Bullet Physics and have referred to the BasicExample provided in the examples folder. I'm attempting a similar integration.

However, when I include #include "btBulletDynamicsCommon.h", I continuously get the linker error shown below. I've been unable to resolve it. Any help would be greatly appreciated.

Environment:

    OS: Ubuntu 24.04

    Qt: 5.15.2

Error:
`
[  0%] Built target MyBullet_autogen_timestamp_deps
[ 16%] Automatic MOC and UIC for target MyBullet
[ 16%] Built target MyBullet_autogen
[ 33%] Building CXX object CMakeFiles/MyBullet.dir/main.cpp.o
[ 50%] Building CXX object CMakeFiles/MyBullet.dir/BulletExample.cpp.o
[ 66%] Linking CXX executable MyBullet
/usr/bin/ld: CMakeFiles/MyBullet.dir/main.cpp.o: in function `main':
/home/workSpace/QtProject/Repo/bulletexperiments/MyBullet/main.cpp:26:(.text+0x22a): undefined reference to `BulletExampleCreateFunc(CommonExampleOptions&)'
/usr/bin/ld: CMakeFiles/MyBullet.dir/BulletExample.cpp.o: in function `btAlignedAllocator<int, 16u>::deallocate(int*)':
/home//workSpace/QtProject/Repo/bullet3/src/LinearMath/btAlignedAllocator.h:97:(.text._ZN18btAlignedAllocatorIiLj16EE10deallocateEPi[_ZN18btAlignedAllocatorIiLj16EE10deallocateEPi]+0x1c): undefined reference to `btAlignedFreeInternal(void*)'
/usr/bin/ld: CMakeFiles/MyBullet.dir/BulletExample.cpp.o: in function `btAlignedAllocator<btHashInt, 16u>::deallocate(btHashInt*)':
/home/workSpace/QtProject/Repo/bullet3/src/LinearMath/btAlignedAllocator.h:97:(.text._ZN18btAlignedAllocatorI9btHashIntLj16EE10deallocateEPS0_[_ZN18btAlignedAllocatorI9btHashIntLj16EE10deallocateEPS0_]+0x1c): undefined reference to `btAlignedFreeInternal(void*)'
/usr/bin/ld: CMakeFiles/MyBullet.dir/BulletExample.cpp.o: in function `btAlignedAllocator<btTriangleInfo, 16u>::deallocate(btTriangleInfo*)':
/home/workSpace/QtProject/Repo/bullet3/src/LinearMath/btAlignedAllocator.h:97:(.text._ZN18btAlignedAllocatorI14btTriangleInfoLj16EE10deallocateEPS0_[_ZN18btAlignedAllocatorI14btTriangleInfoLj16EE10deallocateEPS0_]+0x1c): undefined reference to `btAlignedFreeInternal(void*)'
/usr/bin/ld: CMakeFiles/MyBullet.dir/BulletExample.cpp.o:(.data.rel.ro._ZTV17btTypedConstraint[_ZTV17btTypedConstraint]+0x60): undefined reference to `btTypedConstraint::serialize(void*, btSerializer*) const'
collect2: error: ld returned 1 exit status
gmake[2]: *** [CMakeFiles/MyBullet.dir/build.make:141: MyBullet] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:86: CMakeFiles/MyBullet.dir/all] Error 2
gmake: *** [Makefile:136: all] Error 2
18:09:53: The process "/usr/bin/cmake" exited with code 2.
18:09:53: Error while building/deploying project MyBullet (kit: Qt 5.15.2 GCC 64bit)
18:09:53: When executing step "Build"`

I have added the necessary include directories and library paths in my CMakeLists.txt:

cmake:

`set(BULLET_ROOT "/home/workSpace/QtProject/Repo/bullet3")

# Bullet library paths
set(BULLET_INCLUDE_DIR "${BULLET_ROOT}/src")
set(BULLET_LIB_DIR "${BULLET_ROOT}/build/src")
set(BULLET_LIB_DIR "${BULLET_ROOT}/examples")

# Include Bullet headers
include_directories(${BULLET_INCLUDE_DIR})

if(${QT_VERSION_MAJOR} GREATER_EQUAL 6)
  qt_add_executable(MyBullet
    MANUAL_FINALIZATION
    ${PROJECT_SOURCES}
  )

add_executable(MyBullet
  main.cpp
  BulletExample.cpp
  BulletExample.h
)
# Link Qt libraries
target_link_libraries(MyBullet
  Qt6::Core
  Qt6::Gui
  Qt6::Qml
  Qt6::Quick
  Qt6::Quick3D
  /usr/local/lib/libLinearMath.a
  /usr/local/lib/libBulletCollision.a
  /usr/local/lib/libBulletDynamics.a
)

# Link Bullet static libraries
target_link_libraries(MyBullet
  ${BULLET_LIB_DIR}/build/LinearMath/libLinearMath.a
  ${BULLET_LIB_DIR}/build/BulletCollision/libBulletCollision.a
  ${BULLET_LIB_DIR}/build/BulletDynamics/libBulletDynamics.a
)
`