bulletphysics/bullet3 #1786: pybullet leaks memory while alt is pressed and the gui is in focus

ichumuh, 2018-07-09:
If you run this example and press alt while the gui is in focus, you can see the memory usage go up. It stops if you release the key.
```
#!/usr/bin/env python
import pybullet as p
from time import sleep

import pybullet_data

physicsClient = p.connect(p.GUI)#or p.DIRECT for non-graphical version
p.setAdditionalSearchPath(pybullet_data.getDataPath()) #optionally
p.setGravity(0,0,-10)
planeId = p.loadURDF("plane.urdf")
cubeStartPos = [0,0,1]
cubeStartOrientation = p.getQuaternionFromEuler([0,0,0])
boxId = p.loadURDF("r2d2.urdf",cubeStartPos, cubeStartOrientation)
sleep(100)
p.disconnect()
```
However, if you alt tab to a different window, it will continue to leak until you tab back to the gui and press and release alt.
I'm using pybullet 2.0.9 with python 2.7 on ubuntu 16.06.

erwincoumans, 2018-12-22:
Thanks for the report. Suggestion: release that key!

closed 2018-12-22 by erwincoumans

MattiaRacca, 2021-01-29:
Sorry to revive such an old issue but these leaks are still here. I tested this with:
- python 2.7, 3.6, 3.7, 3.8
- pybullet 3.0.8.
- openGL 2 and 3
They happen with **any** Keyboard event. Basically, keeping a button pressed when the GUI is on focus (e.g. to pan the camera) causes leaks. Any clue what could cause this?

Of course, I can use DIRECT or move the camera from code to avoid the problem, but maybe we can find an actual solution 
Here a simple script I was using to debug this:
```python
#! /usr/bin/env python3

import pybullet as pb
import pybullet_data as pb_data
import math

def setup_panda_environment():
    flags = pb.URDF_ENABLE_CACHED_GRAPHICS_SHAPES

    floor = pb.loadURDF("plane.urdf", basePosition=[0., 0., -0.65],
                        useFixedBase=False)
    table = pb.loadURDF("table/table.urdf", [0.5, 0.25, -0.625],
                        pb.getQuaternionFromEuler([0., 0., 0]),
                        useFixedBase=False, flags=flags)

    # base orientation wrt world frame
    orn = pb.getQuaternionFromEuler([0., 0., - math.pi/2.])
    panda = pb.loadURDF("franka_panda/panda.urdf", [0., 0., 0.], orn,
                        useFixedBase=True, flags=flags)

def state_to_message(state, state_dictionary):
    message = ""
    for flag in state_dictionary:
        if (state & flag) == flag:
            message += str(state_dictionary[flag])
    return message

if __name__ == "__main__":
    pb.connect(pb.GUI)
    pb.setAdditionalSearchPath(pb_data.getDataPath())
    setup_panda_environment()

    keyboard_states = {
        pb.KEY_WAS_TRIGGERED: "Pressed",
        pb.KEY_IS_DOWN: "Down",
        pb.KEY_WAS_RELEASED: "Released"
    }

    keyboard_events = {
        pb.B3G_ALT: "ALT",
        pb.B3G_CONTROL: "CTLR",
        pb.B3G_RIGHT_ARROW: "RIGHT",
        pb.B3G_LEFT_ARROW: "LEFT",
        pb.B3G_SHIFT: "SHIFT",
        pb.B3G_HOME: "HOME"
    }

    while(True):
        user_input = input('Logging key events [q to quit]...')

        if user_input == "q":
            break

        ke = pb.getKeyboardEvents()
        for key in ke:
            print('{}: {}'.format(keyboard_events.get(key, key),
                                  state_to_message(ke[key], keyboard_states)))

    pb.disconnect()
```