bulletphysics/bullet3 #1789: Collision between two signed-distance-field objects

rstrudel, 2018-07-12:
Hi,

I would like to use .cdf files as collision shapes for the meshes I load. 
Collision between a static mesh using CDF and non-static meshes using VHACD as collision shape works well (as in the example with duck, spheres and a concave box).

However it seems that collision between two meshes using CDF is not handled . Here is a simple piece of code reproducing the error using the `concave_box` mesh:

`import pybullet as pb
from time import sleep

pb.connect(pb.GUI)
pb.setPhysicsEngineParameter(numSolverIterations=10)
pb.setTimeStep(1./120.)
pb.setGravity(0, 0, -10)

pb.loadURDF("toys/concave_box.urdf",[0,0,2])
pb.loadURDF("toys/concave_box.urdf",[0,0,0],useFixedBase=True)

while 1:
    sleep(0.01) # Time in seconds.
    pb.stepSimulation()`

erwincoumans, 2019-02-08:
This is not implemented at the moment. We would need to provide some way get surface points on one SDF against the other. I suspect there won't be anyone working on this, so contributions are welcome.

SirNate0, 2020-05-26:
I know nothing about the feature (in fact, I didn't know Bullet even offered signed distance field objects until I came across this issue), but wouldn't something like checking for the intersection of the two signed distance fields be simple enough, something like just computing the min between the two of them and sampling over the regions within them that might intersect (based on a bounding box or convex hull or something)? As I said, though, I'm not familiar with the feature at all, and I don't know if that approach would be suitable for per-step calculations nor whether it would be amenable to some sort of up-front cacheable calculation like finding surface points could be.

closed 2021-04-26 by erwincoumans