bulletphysics/bullet3 #1794: Bullet3 CMake BUILED_SHARED_LIBS option on Windows

bitminer49er, 2018-07-18:
On windows what is the CMake BUILED_SHARED_LIBS option supposed to do? Right now it generates gobs of linker errors:

Error	101	error LNK2019: unresolved external symbol __WSAFDIsSet referenced in function enet_socket_wait	C:\projects\bullet\bullet3\build\Extras\BulletRobotics\win32.obj	BulletRobotics
Error	140	error LNK2019: unresolved external symbol __imp_WSAStartup referenced in function enet_initialize	C:\projects\bullet\bullet3\build\Extras\BulletRobotics\win32.obj	BulletRobotics
Error	146	error LNK2019: unresolved external symbol __imp_WSASendTo referenced in function enet_socket_send	C:\projects\bullet\bullet3\build\Extras\BulletRobotics\win32.obj	BulletRobotics
Error	145	error LNK2019: unresolved external symbol __imp_WSARecvFrom referenced in function enet_socket_receive	C:\projects\bullet\bullet3\build\Extras\BulletRobotics\win32.obj	BulletRobotics
Error	143	error LNK2019: unresolved external symbol __imp_WSAGetLastError referenced in function enet_socket_connect	C:\projects\bullet\bullet3\build\Extras\BulletRobotics\win32.obj	BulletRobotics
Error	142	error LNK2019: unresolved external symbol __imp_WSACleanup referenced in function enet_deinitialize	C:\projects\bullet\bullet3\build\Extras\BulletRobotics\win32.obj	BulletRobotics
Error	97	error LNK2019: unresolved external symbol __imp_timeGetTime referenced in function enet_time_get	C:\projects\bullet\bullet3\build\Extras\BulletRobotics\win32.obj	BulletRobotics
Error	100	error LNK2019: unresolved external symbol __imp_timeEndPeriod referenced in function enet_deinitialize	C:\projects\bullet\bullet3\build\Extras\BulletRobotics\win32.obj	BulletRobotics
Error	99	error LNK2019: unresolved external symbol __imp_timeBeginPeriod referenced in function enet_initialize	C:\projects\bullet\bullet3\build\Extras\BulletRobotics\win32.obj	BulletRobotics
Error	135	error LNK2019: unresolved external symbol __imp_socket referenced in function enet_socket_create	C:\projects\bullet\bullet3\build\Extras\BulletRobotics\win32.obj	BulletRobotics
Error	133	error LNK2019: unresolved external symbol __imp_shutdown referenced in function enet_socket_shutdown	C:\projects\bullet\bullet3\build\Extras\BulletRobotics\win32.obj	BulletRobotics
Error	130	error LNK2019: unresolved external symbol __imp_setsockopt referenced in function enet_socket_set_option	C:\projects\bullet\bullet3\build\Extras\BulletRobotics\win32.obj	BulletRobotics
Error	158	error LNK2019: unresolved external symbol __imp_sendto referenced in function "public: virtual int __cdecl CSimpleSocket::Send(unsigned char const *,unsigned __int64)" (?Send@CSimpleSocket@@UEAAHPEBE_K@Z)	C:\projects\bullet\bullet3\build\Extras\BulletRobotics\SimpleSocket.obj	BulletRobotics
Error	157	error LNK2019: unresolved external symbol __imp_send referenced in function "public: virtual int __cdecl CSimpleSocket::Send(unsigned char const *,unsigned __int64)" (?Send@CSimpleSocket@@UEAAHPEBE_K@Z)	C:\projects\bullet\bullet3\build\Extras\BulletRobotics\SimpleSocket.obj	BulletRobotics
Error	128	error LNK2019: unresolved external symbol __imp_select referenced in function enet_socket_wait	C:\projects\bullet\bullet3\build\Extras\BulletRobotics\win32.obj	BulletRobotics
Error	156	error LNK2019: unresolved external symbol __imp_recvfrom referenced in function "public: virtual int __cdecl CSimpleSocket::Receive(int,unsigned char *)" (?Receive@CSimpleSocket@@UEAAHHPEAE@Z)	C:\projects\bullet\bullet3\build\Extras\BulletRobotics\SimpleSocket.obj	BulletRobotics
Error	155	error LNK2019: unresolved external symbol __imp_recv referenced in function "public: virtual int __cdecl CSimpleSocket::Receive(int,unsigned char *)" (?Receive@CSimpleSocket@@UEAAHHPEAE@Z)	C:\projects\bullet\bullet3\build\Extras\BulletRobotics\SimpleSocket.obj	BulletRobotics
Error	125	error LNK2019: unresolved external symbol __imp_ntohs referenced in function enet_socket_accept	C:\projects\bullet\bullet3\build\Extras\BulletRobotics\win32.obj	BulletRobotics
Error	152	error LNK2019: unresolved external symbol __imp_ntohl referenced in function enet_peer_send	C:\projects\bullet\bullet3\build\Extras\BulletRobotics\peer.obj	BulletRobotics
Error	123	error LNK2019: unresolved external symbol __imp_listen referenced in function enet_socket_listen	C:\projects\bullet\bullet3\build\Extras\BulletRobotics\win32.obj	BulletRobotics
Error	111	error LNK2019: unresolved external symbol __imp_ioctlsocket referenced in function enet_socket_set_option	C:\projects\bullet\bullet3\build\Extras\BulletRobotics\win32.obj	BulletRobotics
Error	122	error LNK2019: unresolved external symbol __imp_inet_ntoa referenced in function enet_address_get_host_ip	C:\projects\bullet\bullet3\build\Extras\BulletRobotics\win32.obj	BulletRobotics
Error	119	error LNK2019: unresolved external symbol __imp_inet_addr referenced in function enet_address_set_host	C:\projects\bullet\bullet3\build\Extras\BulletRobotics\win32.obj	BulletRobotics
Error	113	error LNK2019: unresolved external symbol __imp_htons referenced in function "private: bool __cdecl CActiveSocket::ConnectRAW(char const *,unsigned short)" (?ConnectRAW@CActiveSocket@@AEAA_NPEBDG@Z)	C:\projects\bullet\bullet3\build\Extras\BulletRobotics\ActiveSocket.obj	BulletRobotics
Error	154	error LNK2019: unresolved external symbol __imp_getsockopt referenced in function "private: bool __cdecl CSimpleSocket::Flush(void)" (?Flush@CSimpleSocket@@AEAA_NXZ)	C:\projects\bullet\bullet3\build\Extras\BulletRobotics\SimpleSocket.obj	BulletRobotics
Error	162	error LNK2019: unresolved external symbol __imp_getsockname referenced in function "public: virtual bool __cdecl CActiveSocket::Open(char const *,unsigned short)" (?Open@CActiveSocket@@UEAA_NPEBDG@Z)	C:\projects\bullet\bullet3\build\Extras\BulletRobotics\ActiveSocket.obj	BulletRobotics
Error	160	error LNK2019: unresolved external symbol __imp_getpeername referenced in function "public: virtual bool __cdecl CActiveSocket::Open(char const *,unsigned short)" (?Open@CActiveSocket@@UEAA_NPEBDG@Z)	C:\projects\bullet\bullet3\build\Extras\BulletRobotics\ActiveSocket.obj	BulletRobotics
Error	138	error LNK2019: unresolved external symbol __imp_gethostbyname referenced in function enet_address_set_host	C:\projects\bullet\bullet3\build\Extras\BulletRobotics\win32.obj	BulletRobotics
Error	137	error LNK2019: unresolved external symbol __imp_gethostbyaddr referenced in function enet_address_get_host	C:\projects\bullet\bullet3\build\Extras\BulletRobotics\win32.obj	BulletRobotics
Error	109	error LNK2019: unresolved external symbol __imp_connect referenced in function enet_socket_connect	C:\projects\bullet\bullet3\build\Extras\BulletRobotics\win32.obj	BulletRobotics
Error	107	error LNK2019: unresolved external symbol __imp_closesocket referenced in function enet_socket_destroy	C:\projects\bullet\bullet3\build\Extras\BulletRobotics\win32.obj	BulletRobotics
Error	105	error LNK2019: unresolved external symbol __imp_bind referenced in function enet_socket_bind	C:\projects\bullet\bullet3\build\Extras\BulletRobotics\win32.obj	BulletRobotics
Error	103	error LNK2019: unresolved external symbol __imp_accept referenced in function enet_socket_accept	C:\projects\bullet\bullet3\build\Extras\BulletRobotics\win32.obj	BulletRobotics
Error	248	error LNK2019: unresolved external symbol "void __cdecl urdfStringSplit(class btAlignedObjectArray<class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > > &,class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > const &,class btAlignedObjectArray<class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > > const &)" (?urdfStringSplit@@YAXAEAV?$btAlignedObjectArray@V?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@@@AEBV?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@AEBV1@@Z) referenced in function "public: void __cdecl PhysicsServerExample::drawUserDebugLines(void)" (?drawUserDebugLines@PhysicsServerExample@@QEAAXXZ)	C:\projects\bullet\bullet3\build\examples\RobotSimulator\PhysicsServerExample.obj	App_RobotSimulator
Error	247	error LNK2019: unresolved external symbol "public: void __cdecl PhysicsServerSharedMemory::syncPhysicsToGraphics(void)" (?syncPhysicsToGraphics@PhysicsServerSharedMemory@@QEAAXXZ) referenced in function "public: virtual void __cdecl PhysicsServerExample::physicsDebugDraw(int)" (?physicsDebugDraw@PhysicsServerExample@@UEAAXH@Z)	C:\projects\bullet\bullet3\build\examples\RobotSimulator\PhysicsServerExample.obj	App_RobotSimulato [... cut ...]
Error	164	error LNK1120: 34 unresolved externals	C:\projects\bullet\bullet3\build\lib\Debug\BulletRobotics_Debug.dll	BulletRobotics
Error	67	error LNK1120: 32 unresolved externals	C:\projects\bullet\bullet3\build\lib\Debug\gwen_Debug.dll	gwen
Error	210	error LNK1120: 28 unresolved externals	C:\projects\bullet\bullet3\build\examples\RobotSimulator\Debug\App_RobotSimulator_NoGUI_d.exe	App_RobotSimulator_NoGUI
Error	180	error LNK1120: 12 unresolved externals	C:\projects\bullet\bullet3\build\examples\RobotSimulator\Debug\App_HelloBulletRobotics_d.exe	App_HelloBulletRobotics

bitminer49er, 2018-07-19:
> BUILD_SHARED_LIBS  is disabled on WIN32:
> SET(BUILD_SHARED_LIBS OFF CACHE BOOL "Shared Libs" FORCE)
>
> How did you enable it?
>
> [
> {
> "@type": "EmailMessage",
> "@type": "ViewAction",
> "target": "https://github.com/bulletphysics/bullet3/issues/1794#issuecomment-406275480",
> "url": "https://github.com/bulletphysics/bullet3/issues/1794#issuecomment-406275480",
> "name": "View Issue"
> },
> "description": "View this Issue on GitHub",
> "publisher": {
> "@type": "Organization",
> "name": "GitHub",
> "url": "https://github.com"
> }
> },
> {
> "@type": "MessageCard",
> "@context": "http://schema.org/extensions",
> "hideOriginalBody": "false",
> "originator": "AF6C5A86-E920-430C-9C59-A73278B5EFEB",
> "title": "Re: [bulletphysics/bullet3] Bullet3 CMake BUILED_SHARED_LIBS option on Windows (#1794)",
> "sections": [
> {
> "text": "",
> "activityTitle": "**erwincoumans**",
> "activityImage": "https://assets-cdn.github.com/images/email/message_cards/avatar.png",
> "activitySubtitle": "@erwincoumans",
> "facts": [
>
> ]
> }
> ],
> {
> "name": "Add a comment",
> "@type": "ActionCard",
> "inputs": [
> {
> "isMultiLine": true,
> "@type": "TextInput",
> "id": "IssueComment",
> "isRequired": false
> }
> ],
> "actions": [
> {
> "name": "Comment",
> "@type": "HttpPOST",
> "target": "https://api.github.com",
> "body": "{\n\"commandName\": \"IssueComment\",\n\"repositoryFullName\": \"bulletphysics/bullet3\",\n\"issueId\": 1794,\n\"IssueComment\": \"{{IssueComment.value}}\"\n}"
> }
> ]
> },
> {
> "targets": [
> {
> "os": "default",
> "uri": "https://github.com/bulletphysics/bullet3/issues/1794#issuecomment-406275480"
> }
> ],
> "@type": "OpenUri",
> "name": "View on GitHub"
> },
> {
> "name": "Unsubscribe",
> "@type": "HttpPOST",
> "target": "https://api.github.com",
> "body": "{\n\"commandName\": \"MuteNotification\",\n\"threadId\": 357947509\n}"
> }
> ],
> "themeColor": "26292E"
> }
> ]

Simply using cmake gui and also setting cmake args in externalptlroject_add in my super build.

Sent from my U.S. Cellular® Smartphone

bitminer49er, 2018-07-24:
Also from my CMakeCache.txt file

goop
...
...
//Set when you want to build pybullet (Python bindings for Bullet)
BUILD_PYBULLET:BOOL=OFF

//Use shared libraries
BUILD_SHARED_LIBS:BOOL=ON

//Build Unit Tests
BUILD_UNIT_TESTS:BOOL=ON

...
more goop

Are you sure your sure?

From CMakeLists.txt

OPTION(BUILD_SHARED_LIBS "Use shared libraries" OFF)

Then later..

	IF(WIN32)
		SET(BUILD_SHARED_LIBS OFF CACHE BOOL "Shared Libs" FORCE)
	ELSE(WIN32)
		SET(BUILD_SHARED_LIBS ON CACHE BOOL "Shared Libs" FORCE)
	ENDIF(WIN32)

Cuz theres this bit about 

IF(BUILD_PYBULLET)
	FIND_PACKAGE(PythonLibs)

yada yada yada

then the bit above about building shared libs

ENDIF(BUILD_PYBULLET)

You guys ever though about what happens when user does not want to build pybullet?

Ya know there are still those C/C++ guys out there.

My pybullet is off in CMakeCache.txt and was default off see above exeerpt from my CMakeCache.txt

Are you sure this is a  Closed Ticket?  I'm a thinkin no.

mattparks, 2018-07-26:
Any reason why BUILD_SHARED_LIBS has to be off on windows? Why can bullet not be built as dlls on MSVC?

erwincoumans, 2018-07-27:
There is no-one who supports BUILD_SHARED_LIBS on Windows. If someone wants to add support for BUILD_SHARED_LIBS, feel free to contribute. Some people use BUILD_SHARED_LIBS on Linux and Mac OSX without issue.

closed 2018-07-27 by erwincoumans