bulletphysics/bullet3 #1800: Memory leak in getClosestPoints?

haudren, 2018-07-23:
I am using pybullet to compute distances between a bunch of objects in a scene, but my RAM gets full very quickly... I tried to profile using `memory_profiler` and it seems that the memory allocated by `getClosestPoints` is never freed.

I wrote the following repro program:
```python
import pybullet as p
import pybullet_data
import os
import numpy as np

nr_ducks = 100
nr_steps = 1000

p.connect(p.DIRECT)

root = pybullet_data.getDataPath()

ducks = [p.loadURDF(os.path.join(root, 'duck_vhacd.urdf')) for i in range(nr_ducks)]

for duck in ducks:
    p.resetBasePositionAndOrientation(duck, 0.05*np.random.rand(3), [0, 0, 0, 1])

for i in range(nr_steps):
    distances = [p.getClosestPoints(ducks[0], duck, 0.01) for duck in ducks[1:]]
    del distances  # Should not even be needed

p.disconnect()
```
The resulting memory usage (`mprof run leak_pybullet.py`):
[image]

erwincoumans, 2018-07-23:
Thanks for the report, we'll check it out.

erwincoumans, 2018-07-25:
Thanks for the report, should be fixed here: https://github.com/bulletphysics/bullet3/pull/1804

closed 2018-08-31 by erwincoumans