bulletphysics/bullet3 #1805: getEulerZYX wrong in singularity

mqnc, 2018-07-26:
some parts in the singularity code must be wrong:
```
// Check that pitch is not at a singularity
if (btFabs(m_el[2].x()) >= 1)
{
        euler_out.yaw = 0;
        euler_out2.yaw = 0;

        // From difference of angles formula
        btScalar delta = btAtan2(m_el[0].x(),m_el[0].z());
        if (m_el[2].x() > 0)  //gimbal locked up
        {
                euler_out.pitch = SIMD_PI / btScalar(2.0);
                euler_out2.pitch = SIMD_PI / btScalar(2.0);
                euler_out.roll = euler_out.pitch + delta;
                euler_out2.roll = euler_out.pitch + delta;
        }
        else // gimbal locked down
        {
                euler_out.pitch = -SIMD_PI / btScalar(2.0);
                euler_out2.pitch = -SIMD_PI / btScalar(2.0);
                euler_out.roll = -euler_out.pitch + delta;
                euler_out2.roll = -euler_out.pitch + delta;
        }
}
```

Here are my conversion functions:

```
btVector3 mat2abc(btMatrix3x3 ori){
	btScalar x,y,z;
	ori.getEulerZYX(x,y,z);
	return btVector3(z,y,x);
}

btMatrix3x3 abc2mat(btVector3 abc){
	btMatrix3x3 result;
	result.setEulerZYX(abc[0], abc[1], abc[2]);
	return result;
}
```

They work fine normally, they only result in completely weird directions in the singularity. Try converting `btMatrix3x3(0,0,1,  0,-1,0,  1,0,0)` back and forth, x will show in the negative direction. Also if you rotate that around the x axis, the result is strange.

erwincoumans, 2018-07-30:
Sure, provide some repro case please.

mqnc, 2018-07-30:
```
auto abc_start = btVector3(0, PI/2.0, PI/4.0);
auto mat1 = abc2mat(abc_start);
auto abc_temp = mat2abc(mat1);
auto mat2 = abc2mat(abc_temp);
```
abc_start and abc_temp don't have to be identical due to redundancy in the singularity but mat1 and mat2 should be equal. However, mat1 is correctly

```
 0  -0.707  0.707
 0   0.707  0.707
-1   0      0
```
but mat2 becomes
```
 0  -0.992  0.124
 0   0.124  0.992
-1   0      0
```

erwincoumans, 2018-07-30:
Good point, here is a PR with a suggested fix, similar to URDF dom headers:
https://github.com/bulletphysics/bullet3/pull/1806

https://github.com/ros/urdfdom_headers/blob/e7e0972a4617b8a5df7a274ea3ba3b92e3895a35/urdf_model/include/urdf_model/pose.h#L103

erwincoumans, 2018-08-31:
The quaternion version should be fine now, please use that one. Let's close the issue. Thanks for the report.

closed 2018-08-31 by erwincoumans