bulletphysics/bullet3 #1807: Memory Leak in PhysicsServerCommandProcessor::processRemoveBodyCommand

ziyuli, 2018-07-27:
I am using bullet to load lots of URDFs in SharedMemory example, however the command removeBody is not fully release the memory. The size of m_collisionShapes and m_meshInterfaces seems keep increasing without removed and released.

erwincoumans, 2018-07-27:
Are you calling resetSimulation now and then? It should release all the memory.

ziyuli, 2018-07-27:
Thanks for your prompt reply!

Actually, I am planning to create a simulation with randomly generating several multibodies into the scene and then removing some of them from the scene in lots of iterations. So I probably need to keep the resetSimulation in the end of program.

The testing program is something like this:
```
#include <string>

#include "bullet/LinearMath/btTransform.h"
#include "bullet/PhysicsClientC_API.h"

using namespace std;

static string husky = "./husky/robot_kuka.urdf";
static string chair = "./chair_1/chair.urdf";
static string apple = "./apple/apple.urdf";
static string cat = "./cat_1/cat.urdf";

static float start_x = 0;
static float start_y = 0;
static float start_z = 0;
static b3PhysicsClientHandle client;

int loadURDF(const string& filename)
{
	b3SharedMemoryCommandHandle command = b3LoadUrdfCommandInit(client, filename.c_str());
	b3LoadUrdfCommandSetStartPosition(command, start_x, start_y, start_z);
	b3LoadUrdfCommandSetStartOrientation(command, 0, 0, 0, 1);
	b3LoadUrdfCommandSetUseFixedBase(command, false);
	b3LoadUrdfCommandSetGlobalScaling(command, 1.0f);
	b3LoadUrdfCommandSetUseMultiBody(command, true);
	b3LoadUrdfCommandSetFlags(command, 0);

	b3SharedMemoryStatusHandle statusHandle = b3SubmitClientCommandAndWaitStatus(client, command);
	if (b3GetStatusType(statusHandle) != CMD_URDF_LOADING_COMPLETED) {
		fprintf(stderr, "Cannot load URDF file '%s'.\n", filename.c_str());
		return -1;
	}

	return b3GetStatusBodyIndex(statusHandle);
}

void removeURDF(const int bullet_handle)
{
	if(bullet_handle < 0)
	{
		fprintf(stderr, "Cannot remove URDF '%i'.\n", bullet_handle);
		return;
	}
	b3SharedMemoryStatusHandle statusHandle;
	statusHandle = b3SubmitClientCommandAndWaitStatus(client, b3InitRemoveBodyCommand(client,bullet_handle));
}

int main()
{
	client = b3ConnectPhysicsDirect();
	
	int it = 0;
	while(it++ < 1000){
		
		int b0 = loadURDF(husky);
		int b1 = loadURDF(chair);
		int b2 = loadURDF(apple);
		int b3 =loadURDF(cat);

		// Do some simulation...		
		printf("Iterations: %d\n", it);

		removeURDF(b0);
		removeURDF(b1);
		removeURDF(b2);
		removeURDF(b3);
	}
	
	b3SubmitClientCommandAndWaitStatus(client, b3InitResetSimulationCommand(client));
	return 0;
}
```

After 1000 iterations, it occupies approx. 1GB of RAM.

erwincoumans, 2018-08-16:
This is not supported, you need to periodically call resetSimulation, since memory is only freed at that point.

closed 2018-08-16 by erwincoumans