bulletphysics/bullet3 #1810: retrieve type of btTriangleCallback used when btConcaveShape::processAllTriangles() is called

gjaegy, 2018-08-01:
Hi,

I have created a custom class derived from btConcaveShape() to handle the collision with the triangles of my terrain mesh. This works great.

In order to complete an optimization, I would need to access the other collision shape that is currently being tested against my derived btConcaveShape() (actually I need the full AABB of the other object FYI).

That would be quite easy to do if I had some information about the current callback type being used. I could cast the callback to the specialized type (for instance btConvexTriangleCallback, or BridgeTriangleConvexcastCallback) and then get the relevant information I need.

However, there is no callback type being transfered, so I have currently no way of casting the callback to the specialized callback type to retrieve the information I need.

One solution would be to add a pure virtual getType() method to the btTriangleCallback interface and returns a different type in the derived class implementation.

I could do that change if needed, but would you merge such a pull request ?

Cheers
Greg

gjaegy, 2018-08-01:
Another option would be to pass down the two colliding rigid bodies as parameters to the processAllTriangles() method...

BluePrintRandom, 2018-08-01:
If the shape is not changing {the colliding object not the terrain} one can
build a bvhtree of the object on frame 0 and then on collision get the
closest triangle [nearest] or closest triangles [findRange] almost instanly

{this is great for dynamic destruction}

gjaegy, 2018-08-01:
Thanks for your answer. I am not sure I understand the relation with my issue however.. I would need to access some information about the other shape colliding in my processAllTriangles() callback...

gjaegy, 2018-08-02:
submitted a pull request

closed 2018-08-02 by gjaegy