bulletphysics/bullet3 #1815: Bug: joint damping in URDF causes undesired behaviors.

bdelhaisse, 2018-08-03:
Hi,

I have an undesired behavior when setting the joint damping to different values for any joint in the URDF. This doesn't happen in Gazebo.

Here are some samples to better show the problem: 

* when setting the joint damping to 0, everything works fine:
[image]

* when setting the joint damping of one joint to 2, the undesired behavior is delayed:
[image]

* when setting the joint damping to 3 or higher:
[image]

Also, is there a method that allows the user to change joint informations in an online fashion in pybullet (once the URDF has been loaded)? Currently, if I want to change for instance the joint friction or joint limits, I have to stop the simulation, update the URDF, and then rerun pybullet, which is not practical.

Thanks!

erwincoumans, 2018-08-16:
The joint damping values need to be really small probably (I haven't tested the feature). Also, we didn't expose modifying the joint limits at run-time yet, it should be reasonably easy to modify them with some code changes. See the [PyBullet quickstart guide](https://docs.google.com/document/d/10sXEhzFRSnvFcl3XxNGhnD4N2SedqwdAvK3dsihxVUA/edit#heading=h.2ye70wns7io3) for things that can be modified through APIs.

closed 2019-02-08 by erwincoumans

josiahls, 2019-04-06:
Why do we need to modify the damping at runtime? I downloaded a fetch robot URDF from the Fetch Robotics github with damping that ranges from 1 - 100 in the joint tags. Setting the damping to 0 makes the robot act like a rag-doll but at least doesn't explode.

I am getting a similar issue to @bdelhaisse where the robot basically explodes / joints start spinning extremely fast. Is there a fundamental difference between bullet3 and other physics engines in regards to damping? [Fetch URDF description](https://github.com/fetchrobotics/fetch_ros/blob/melodic-devel/fetch_description/robots/fetch.urdf)

erwincoumans, 2019-04-06:
We only interpret the basic URDF settings and the meaning of the constant used for joint damping is specific to the implementation. I would recommend using changeDynamics to set the joint damping to a small value.