bulletphysics/bullet3 #1817: Bug: small vibrations when clicking with the mouse

bdelhaisse, 2018-08-03:
Hi,

I have loaded a robotic hand in pybullet. When setting the joint positions to some values, and letting the simulator run, the robot shakes as depicted below:
[image]

Next, by clicking with the left button on the mouse in the world, I have some joints that jump. This is depicted below (everytime the thumb jumps, it is because I clicked):
[image]
I don't have this when clicking with the scroll or right button.

I was wondering what cause these behaviors and if there would be a way to solve them.

Thanks.

erwincoumans, 2018-08-16:
Can you share the URDF/PyBullet file to see if we can reproduce the problem? Also, we fixed some issues recently, are you using the very latest version of PyBullet? pybullet 2.1.2 or later?

erwincoumans, 2018-08-31:
It is likely due to a constraint added between a fixed body and the fixed world. We should avoid adding such constraint.

erwincoumans, 2018-12-22:
We cannot take action without a small reproduction case. Let's close it.

closed 2018-12-22 by erwincoumans

v-wewei, 2021-01-23:
Hi, @erwincoumans,  i meet the same problem recently. when i create some constriants between two joints for each finger in  a robotics hand. Some of the joint show unresonable behaviour. When i click on my hand, the ring (fourth) finger got shaked at same time, corresponding joint index pair is (16, 17)
the code is shown as blew.
    cid_thumb = p.createConstraint(hithandId, 4, hithandId, 5, jointType=p.JOINT_GEAR,
                                   jointAxis=[1, 0, 0],
                                   parentFramePosition=[0, 0, 0],
                                   childFramePosition=[0, 0, 0],
                                   )
    p.changeConstraint(cid_thumb, gearRatio=-1.0, erp=2.0, maxForce=1500)
   
    cid_fore = p.createConstraint(hithandId, 8, hithandId, 9, jointType=p.JOINT_GEAR,
                                  jointAxis=[1, 0, 0],
                                  parentFramePosition=[0, 0, 0],
                                  childFramePosition=[0, 0, 0],
                                  )
    p.changeConstraint(cid_fore, gearRatio=-1.0, erp=2.0, maxForce=1500)
  
    cid_middle = p.createConstraint(hithandId, 12, hithandId, 13, jointType=p.JOINT_GEAR,
                                    jointAxis=[1, 0, 0],
                                    parentFramePosition=[0, 0, 0],
                                    childFramePosition=[0, 0, 0],
                                    )
    p.changeConstraint(cid_middle, gearRatio=-1.0, erp=2.0, maxForce=1500)
    
    cid_ring = p.createConstraint(hithandId, 16, hithandId, 17, jointType=p.JOINT_GEAR,
                                  jointAxis=[1, 0, 0],
                                  parentFramePosition=[0, 0, 0],
                                  childFramePosition=[0, 0, 0],
                                  )
    p.changeConstraint(cid_ring, gearRatio=-1.0, erp=2.0, maxForce=1500)
    
    cid_little = p.createConstraint(hithandId, 20, hithandId, 21, jointType=p.JOINT_GEAR,
                                    jointAxis=[1, 0, 0],
                                    parentFramePosition=[0, 0, 0],
                                    childFramePosition=[0, 0, 0],
                                    )
    p.changeConstraint(cid_little, gearRatio=-1.0, erp=2.0, maxForce=1500)

but if only create costraint for joint  (16, 17) , it works.