bulletphysics/bullet3 #1821: A number of URDFs specify the invalid "0.0" version

gkjohnson, 2018-08-04:
A number of the URDF files in the `/data` folder declare the XML version to be `0.0`. See here:

https://github.com/bulletphysics/bullet3/blob/master/data/cube_concave.urdf#L1

There are quite a few others, too. I was trying to parse the xml in Javascript and got the following errors when using [DOMParser](https://developer.mozilla.org/en-US/docs/Web/API/DOMParser):

```xml
<h3>This page contains the following errors:</h3>
<div style="font-family:monospace;font-size:12px">
  error on line 1 at column 20: Unsupported version '0.0'
</div>
```

erwincoumans, 2018-08-16:
PyBullet should handle those xml files anyway. If you use a program that doesn't handle them and want to fix the URDF files, feel free to submit a pull request.

closed 2018-08-16 by erwincoumans

gkjohnson, 2018-08-16:
I am using them in a different application -- I'll make a PR. Thanks!