bulletphysics/bullet3 #1835: How to get the simulation time at an instant?

abhiksingla, 2018-08-13:
I need assistance to know the simulation time spend at any instant. I understand that sim time should be different from the wall clock time but unable to get the measure.

erwincoumans, 2018-08-16:
You process the simulation using stepSimulation(deltaTime), so you should be able to track some notion of time.

closed 2018-08-16 by erwincoumans

rohit-kumar-j, 2021-01-10:
@erwincoumans  I think the following might be a workaround. Is there a way to access the internal time_stamp of the simulation and print it like below?

```
import time
import pybullet as p

dummy = 0
time_step=1/240
p.setTimeStep(time_step)
p.setRealTimeSimulation(1)

while 1:  
	p.stepSimulation()
	dummy = dummy + 1
	if dummy == 1 :
		delay = time.time()
		time.sleep(1)
		print('delay')
		print('time starts now!')
	#else: print('no delay')
	
	print(time.time()-delay)

p.setRealTimeSimulation(0)

```

These are the values that I get (real-time values not internal time_stamp):

```
delay
time starts now!
1.0009171962738037
1.119473934173584
1.226822853088379
1.3307716846466064
1.4232892990112305
1.5069279670715332
1.6108765602111816
1.7392501831054688
1.8484127521514893
1.9666316509246826
2.0784430503845215
2.1850509643554688
2.286367177963257
2.3844828605651855
2.4708166122436523
2.5747876167297363
2.678741693496704
2.79878306388855
2.8956520557403564
2.9949910640716553
3.098883628845215
3.202812671661377
3.308270215988159
3.4229214191436768
```

rohit-kumar-j, 2021-01-17:
(Hopefully)...Improved Version: (Directly from [inverse_kinematics.py](https://github.com/bulletphysics/bullet3/blob/master/examples/pybullet/examples/inverse_kinematics.py)) -- line 54

```
i=0
while 1:
  i+=1
  if (useRealTimeSimulation):
    dt = datetime.now()
    t = (dt.second / 60.) * 2. * math.pi
  else:
    t = t + 0.01

  if (useSimulation and useRealTimeSimulation == 0):
    p.stepSimulation()
```