bulletphysics/bullet3 #1836: SegFault when training RL agent using camera

btiddies, 2018-08-13:
Hello, 

I am trying to train an RL agent that has a body mounted camera. I am using MPI to run several processes. It works well without a camera input, but after a variable number of steps I will get a segfault:

[brendan:05577] *** Process received signal ***
[brendan:05577] Signal: Segmentation fault (11)
[brendan:05577] Signal code: Address not mapped (1)
[brendan:05577] Failing at address: 0xfffffffe02660ad0
[brendan:05577] [ 0] /lib/x86_64-linux-gnu/libpthread.so.0(+0x11390)[0x7f258a516390]
[brendan:05577] [ 1] /usr/local/lib/python3.5/dist-packages/pybullet.cpython-35m-x86_64-linux-gnu.so(_ZN6Shader8fragmentE3vecILm3EfER8TGAColor+0x175)[0x7f25889ae815]
[brendan:05577] [ 2] /usr/local/lib/python3.5/dist-packages/pybullet.cpython-35m-x86_64-linux-gnu.so(_Z15triangleClippedR3matILm4ELm3EfES1_R7IShaderR8TGAImagePfPiRKS_ILm4ELm4EfEi+0x7c9)[0x7f25889a8659]
[brendan:05577] [ 3] /usr/local/lib/python3.5/dist-packages/pybullet.cpython-35m-x86_64-linux-gnu.so(_ZN12TinyRenderer12renderObjectER20TinyRenderObjectData+0x1b3f)[0x7f25889ac8ff]
[brendan:05577] [ 4] /usr/local/lib/python3.5/dist-packages/pybullet.cpython-35m-x86_64-linux-gnu.so(_ZN32TinyRendererVisualShapeConverter6renderEPKfS1_+0x768)[0x7f2588a2cf98]
[brendan:05577] [ 5] /usr/local/lib/python3.5/dist-packages/pybullet.cpython-35m-x86_64-linux-gnu.so(_ZN29PhysicsServerCommandProcessor32processRequestCameraImageCommandERK19SharedMemoryCommandR18SharedMemoryStatusPci+0x2db)[0x7f25889d911b]
[brendan:05577] [ 6] /usr/local/lib/python3.5/dist-packages/pybullet.cpython-35m-x86_64-linux-gnu.so(_ZN29PhysicsServerCommandProcessor14processCommandERK19SharedMemoryCommandR18SharedMemoryStatusPci+0x4de)[0x7f25889fa1ce]
[brendan:05577] [ 7] /usr/local/lib/python3.5/dist-packages/pybullet.cpython-35m-x86_64-linux-gnu.so(_ZN13PhysicsDirect13processCameraERK19SharedMemoryCommand+0x76)[0x7f25889cd716]
[brendan:05577] [ 8] /usr/local/lib/python3.5/dist-packages/pybullet.cpython-35m-x86_64-linux-gnu.so(b3SubmitClientCommand+0x14)[0x7f2588a1f864]
[brendan:05577] [ 9] /usr/local/lib/python3.5/dist-packages/pybullet.cpython-35m-x86_64-linux-gnu.so(b3SubmitClientCommandAndWaitStatus+0x80)[0x7f2588a1f900]
[brendan:05577] [10] /usr/local/lib/python3.5/dist-packages/pybullet.cpython-35m-x86_64-linux-gnu.so(+0x17a6eb)[0x7f25889956eb]
[brendan:05577] [11] python3(PyCFunction_Call+0x77)[0x4e9ba7]
[brendan:05577] [12] python3(PyEval_EvalFrameEx+0x59f5)[0x53c6d5]
[brendan:05577] [13] python3(PyEval_EvalFrameEx+0x4b04)[0x53b7e4]
[brendan:05577] [14] python3[0x540199]
[brendan:05577] [15] python3(PyEval_EvalCode+0x1f)[0x540e4f]
[brendan:05577] [16] python3[0x60c272]
[brendan:05577] [17] python3(PyRun_FileExFlags+0x9a)[0x60e71a]
[brendan:05577] [18] python3(PyRun_SimpleFileExFlags+0x1bc)[0x60ef0c]
[brendan:05577] [19] python3(Py_Main+0x456)[0x63fb26]
[brendan:05577] [20] python3(main+0xe1)[0x4cfeb1]
[brendan:05577] [21] /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xf0)[0x7f258a15b830]
[brendan:05577] [22] python3(_start+0x29)[0x5d6049]
[brendan:05577] *** End of error message ***

You can find a simple demonstration at the following repo:

https://bitbucket.org/brendantidd/pybullet_test/src/master/

Run with:

mpirun -np 6 python3 run.py

I'm not sure if the issue is related to 
https://github.com/bulletphysics/bullet3/issues/1295 

I am assuming that memory for the tiny renderer is being shared among processes, but I am unsure of how to fix this. Any help would be appreciated!

Kind regards

Brendan

erwincoumans, 2018-08-16:
It is very odd, since each process should run isolated and TinyRenderer is just a CPU renderer. Are you using DIRECT mode for all pybullet instances?

btiddies, 2018-08-16:
yes, DIRECT mode for all instances. I can confirm that without the image code it does not segfault. I have tried reducing how frequently I get an image, and even limiting one process getting an image at a time, which possibly delays the segfault, but still occurs.

On a side note, I just wanted to say that I have been very interested in your paper recently using the minitaur transferring from sim to real, awesome stuff! I am hoping this could be the direction of my work someday. 

Brendan

neveryoubetray, 2018-08-22:
It seams that p.computeProjectionMatrixFOV() could be the culprit (at least when I run the could without it I haven't been able to break it yet). 

I have tried running p.computeProjectionMatrix() instead, though haven't been able to get it do what I want (and I'm not sure if it would fix my issue). The docs list the following arguments, how can I work them out? There doesn't seem to be an example of usage of p.computeProjectionMatrix().

left screen (canvas) coordinate
right screen (canvas) coordinate
bottom screen (canvas) coordinate
top screen (canvas) coordinate

I am trying to use camera that moves with the body, if anyone knows a a better way to get this I would appreciate it.

Kind regards 

Brendan

btiddies, 2018-08-22:
It seams that p.computeProjectionMatrixFOV() could be the culprit (at least when I run the could without it I haven't been able to break it yet). 

I have tried running p.computeProjectionMatrix() instead, though haven't been able to get it do what I want (and I'm not sure if it would fix my issue). The docs list the following arguments, how can I work them out? There doesn't seem to be an example of usage of p.computeProjectionMatrix().

left screen (canvas) coordinate
right screen (canvas) coordinate
bottom screen (canvas) coordinate
top screen (canvas) coordinate

I am trying to use camera that moves with the body, if anyone knows a a better way to get this I would appreciate it.

Kind regards 

Brendan

erwincoumans, 2018-08-25:
You should not use the computeProjectionMatrixFOV to move the camera.

Use the view matrix, or even easier:
pybullet.resetDebugVisualizerCamera(distance,yaw,pitch,targetPos);

closed 2018-08-25 by erwincoumans

btiddies, 2018-08-26:
Thank you for your reply, just a couple of quick questions.

Did you mean just using viewMatrix in the call to getCameraImage()? I have tried getCameraImage() just using the viewMatrix input, but the image position doesn't change unless there is a projectionMatrix as well:

// Image position changes:
vm = p.computeViewMatrix([0,0,1], [1,0,1], [0,0,1])
pm = p.computeProjectionMatrixFOV(60, 1, 0.1, 1000);
image = p.getCameraImage(32, 32, viewMatrix=vm, projectionMatrix=pm, renderer=p.ER_TINY_RENDERER)

// Image position doesn't change
vm = p.computeViewMatrix([0,0,1], [1,0,1], [0,0,1])
image = p.getCameraImage(32, 32, viewMatrix=vm, renderer=p.ER_TINY_RENDERER)

Is a way to set the projectionMatrix once (given that it doesn't change)? It reverts back to default once I call getCameraImage() without the projectionMatrix argument. 

Using p.resetDebugVisualizerCamera() is a good alternative, though doesn't allow me to capture the roll of the agent, which I would ideally like (and trivially requires an additional call to getCameraImage() for debugging/seeing what the agent is doing externally).

Thanks for your time Erwin.

btiddies, 2018-08-29:
So just to double check, getDebugVisualizerCamera will only work if in GUI mode? (and similarly if I try the resetDebugVisualizerCamera option). I saved a projection matrix from getDebugVisualizerCamera I got from GUI mode to use in DIRECT mode, then call something like:

vm = p.computeViewMatrix([0,0,1], [1,0,1], [0,0,1])
image = p.getCameraImage(32, 32, viewMatrix=vm, projectionMatrix=savedPM, renderer=p.ER_TINY_RENDERER)

I am still getting segfaults. To minimise the occurrence I have reduced the number of process from 16 to 8, and run 3 of my vision based experiments at the same time. 1-2 will segfault in an hour or so of training, but usually one will survive for a few hours (at this stage it is enough time to validate my experiment). For now this is a partial solution.

Thanks again for your time.