bulletphysics/bullet3 #1837: pip3 fails to install pybullet on macOS High Sierra 10.13.5

MatchesMalonee, 2018-08-15:
I get this error when I try to install the pybullet library via `pip3` on macOS HS.
```
src/LinearMath/btQuickprof.cpp:713:10: error: thread-local storage is not supported for the current target
    static __thread unsigned int sThreadIndex = kNullIndex;
           ^
  5 warnings and 1 error generated.
  error: command '/usr/bin/clang' failed with exit status 1
  
  ----------------------------------------
  Failed building wheel for pybullet
  Running setup.py clean for pybullet
Failed to build pybullet
Installing collected packages: pybullet
  Running setup.py install for pybullet ... /
```

I'm running Python 3.6.4, no Anaconda, and clang version `Apple LLVM version 9.0.0 (clang-900.0.39.2)`

MatchesMalonee, 2018-08-15:
I also noticed that most of the files in `src/LinearMath/` directory are at least 7-8 months old, except for three files that were 16 days old as of writing this comment. `btQuickprof.cpp` is one of those files.

erwincoumans, 2018-08-16:
pip3 install pybullet works fine here for all platforms, including Mac OSX High Sierra 10.13.5. If you figure out what the problem is, please let us know.

MatchesMalonee, 2018-08-18:
Is the standalone python 3.6 environment not supported by default?

erwincoumans, 2018-08-18:
There is no support, volunteers help make things work. pybullet compiles and runs for the platforms and configurations , I've not run into those Mac/Python issues myself, and don't have time tracking down everyones issues. If you find a solution, please consider sharing a PR.

erwincoumans, 2018-09-14:
It must be some clang/Xcode/compiler issue.

Xcode 9.4.1, Build version 9F2000, can you try updating to a newer Xcode / clang version?

erwincoumans, 2018-09-14:
Some others had similar issues, it is not clear what is causing it and we have no bandwidth on solving it for you.
https://github.com/bulletphysics/bullet3/issues/830
https://github.com/bulletphysics/bullet3/issues/1207

closed 2018-09-14 by erwincoumans