bulletphysics/bullet3 #1841: Pybullet Output when connecting / disconnecting

MichaelLutter, 2018-08-21:
Can you remove / comment the output when connecting / disconnecting to the bullet simulator. With the newest Pip Pybullet Version 2.1.2, bullet prints "hi!" when connecting and "bye!" when disconnecting. 

Finding the origin of the mystical "hi!" has cost me a few hours and made me nervously search all my code.  Can you also use the word Pybullet in the prints, if you insert new print statement in future revisions?

erwincoumans, 2018-08-22:
Yes, it is a debug printf in the upcoming collisionFilterPlugin, I'll remove it.

https://github.com/bulletphysics/bullet3/blob/master/examples/SharedMemory/plugins/collisionFilterPlugin/collisionFilterPlugin.cpp

Thanks for the report!

erwincoumans, 2018-08-31:
It should be fixed by now. Thanks for the report!

closed 2018-08-31 by erwincoumans