bulletphysics/bullet3 #1845: pybullet set initial velocity of urdf object

aznshodan, 2018-08-25:
is there a way I can set initial velocity of a urdf object?
For instance, I have a sphere2.urdf. I'd like to set some initial velocity to this so that when the simulator starts, it is already moving at the velocity I defined.

erwincoumans, 2018-08-25:
You can use resetBaseVelocity for that. See the PyBullet quickstart guide:
https://docs.google.com/document/d/10sXEhzFRSnvFcl3XxNGhnD4N2SedqwdAvK3dsihxVUA/edit#

closed 2018-08-25 by erwincoumans

arpastrana, 2019-02-14:
Hi @erwincoumans. Does using `p.resetBaseVelocity()` have a similar impact as p.resetBasePositionAndOrientation() in that, if applied during a running simulation, it will override the effect of all physics simulation?

erwincoumans, 2019-02-14:
Yes, it will override the physics calculation, so those reset* functions are best used before stepping the simulation.

arpastrana, 2019-02-14:
👍