bulletphysics/bullet3 #1864: Failed building wheel for pybullet

JobSmith, 2018-09-08:
I failed to install pybullet on win10:
[image]

erwincoumans, 2018-09-08:
Never seen that error. You can also simply do
~~~~~~~~~~~~
git clone https://github.com/bulletphysics/bullet3
or download and unzip https://github.com/bulletphysics/bullet3/archive/master.zip and then:
cd bullet3
python setup.py build
python setup.py install
~~~~~~~~~~~~

erwincoumans, 2018-09-08:
[image]

erwincoumans, 2018-09-08:
It could be Windows anaconda related? Although on Mac, we use Anaconda without issue.

JobSmith, 2018-09-08:
This command also occured a ERROR.
```
python setup.py build
```
[image]

closed 2018-09-10 by JobSmith

Mahbub-A-Rob, 2018-12-04:
I had the similar issue on Windows 10, Anaconda. I created a virtual environment with python version 3.5.3, then installed pybullet. 

    conda create --name pybulletenv python=3.5.3
    conda activate 
    pip install pybullet

franferri, 2020-04-04:
The error happens to me, Anaconda Powershell Prompt, brand new installation (I didn't had any anaconda installation befor):

conda create -n rex python=3.7 anaconda
conda activate rex
pip install pybullet

`
...
    copying examples\pybullet\gym\pybullet_data\xarm\xarm_gripper\meshes\right_inner_knuckle.STL -> build\lib.win-amd64-3.7\pybullet_data\xarm\xarm_gripper\meshes
    copying examples\pybullet\gym\pybullet_data\xarm\xarm_gripper\meshes\right_outer_knuckle.STL -> build\lib.win-amd64-3.7\pybullet_data\xarm\xarm_gripper\meshes
    running build_ext
    building 'pybullet' extension
    error: Microsoft Visual C++ 14.0 is required. Get it with "Build Tools for Visual Studio": https://visualstudio.microsoft.com/downloads/
    ----------------------------------------
ERROR: Command errored out with exit status 1: 'C:\Users\fran\anaconda3\python.exe' -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'C:\\Users\\fran\\AppData\\Local\\Temp\\pip-install-na4m0a5v\\pybullet\\setup.py'"'"'; __file__='"'"'C:\\Users\\fran\\AppData\\Local\\Temp\\pip-install-na4m0a5v\\pybullet\\setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record 'C:\Users\fran\AppData\Local\Temp\pip-record-io0biyem\install-record.txt' --single-version-externally-managed --compile --install-headers 'C:\Users\fran\anaconda3\Include\pybullet' Check the logs for full command output.
`

I have installed Microsoft Visual C++ Redistributable for Visual Studio 2019 from here: https://visualstudio.microsoft.com/downloads/#build-tools-for-visual-studio-2017 for 32b and 64b

Any tips?

erwincoumans, 2020-04-04:
I've only used Visual Studio 2019 command prompt and have python installed in c:\python36 and it worked fine. Never used anaconda on Windows, does Anaconda have support forums for Windows/Visual Studio?

franferri, 2020-04-04:
No idea, but now i have installed, explicit 14.0, not 14.25... and it returns:
`
ndows Kits\10\include\10.0.10240.0\ucrt" /Tcexamples/pybullet/pybullet.c /Fobuild\temp.win-amd64-3.7\Release\examples/pybullet/pybullet.obj -DGWEN_COMPILE_STATIC -DBT_USE_DOUBLE_PRECISION -DBT_ENABLE_ENET -DBT_ENABLE_CLSOCKET -DB3_DUMP_PYTHON_VERSION -DEGL_ADD_PYTHON_INIT -DB3_ENABLE_FILEIO_PLUGIN -DB3_USE_ZIPFILE_FILEIO -DBT_THREADSAFE=1 -DSTATIC_LINK_SPD_PLUGIN -DBT_ENABLE_VHACD -DPYBULLET_USE_NUMPY -DWIN32 -DGLEW_STATIC
    pybullet.c
    c:\users\fran\anaconda3\include\pyconfig.h(203): fatal error C1083: Cannot open include file: 'basetsd.h': No such file or directory
    error: command 'C:\\Program Files (x86)\\Microsoft Visual Studio 14.0\\VC\\BIN\\x86_amd64\\cl.exe' failed with exit status 2
    ----------------------------------------
ERROR: Command errored out with exit status 1: 'C:\Users\fran\anaconda3\python.exe' -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'C:\\Users\\fran\\AppData\\Local\\Temp\\pip-install-do_k7tuy\\pybullet\\setup.py'"'"'; __file__='"'"'C:\\Users\\fran\\AppData\\Local\\Temp\\pip-install-do_k7tuy\\pybullet\\setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record 'C:\Users\fran\AppData\Local\Temp\pip-record-atqwrdq5\install-record.txt' --single-version-externally-managed --compile --install-headers 'C:\Users\fran\anaconda3\Include\pybullet' Check the logs for full command output.
`

franferri, 2020-04-04:
I'll try the visualstudio command prompt as you indicate
Thanks!

franferri, 2020-04-04:
Is not pybullet issue, is a known issue: https://stackoverflow.com/questions/23691564/running-cython-in-windows-x64-fatal-error-c1083-cannot-open-include-file-ba/42624713#42624713

EngineerSte, 2020-07-10:
Did a solution ever get found for this? 

I'm trying to install pybullet into my anaconda environment. Currently getting an error "failure to build pybullet wheel".  Also getting "ERROR: Command errrored out with exit status 1"

Seems to happen when building pybullet wheel (setup.py). I made sure my setup.py was updated. 

Also error message suggests I need Visual Studio 2019 14.0. So I install x64 and x86 versions.

os: windows 10 64bit
conda version: 4.8.3
python version: 3.7.7
pip version: 20.1.1

erwincoumans, 2020-07-10:
If you have Python installed in the root of c:\ for example c:\python3.7 and pip and Visual Studio 2019 and use a command-prompt.
pip install pybullet should work on Windows, Linux and Mac OSX.
Alternatively, running python setup.py build or python setup.py develop or python setup.py install should work on all platforms.

EngineerSte, 2020-07-10:
I get the same errors installing from command prompt. I have python3.8.3 installed in c:\users\steph and when I run pip install pybullet here I get the errors. 

 Visual Studio 2019 is installed. I was warned to update pip, so I did, but that had no effect. 

Error happens when setting up the wheel I think. I'm not sure what that means. 

Also I have the error message "error: Microsoft Visual C++ 14.0 is required." I have installed this also. 

I have also tried to manually extract the library using the link you provided above. I made a bullet3 folder in the lib\site-packages folder of my conda env. And then from the conda terminal I ran python setup.py build while in the bullet3 folder. I got the same error about  Microsoft Visual C++ 14.0 being required. Not entirely sure if this is what you were suggesting?

Thank you for the help!

erwincoumans, 2020-07-10:
This is not a regular command prompt. You need the Visual Studio 2019 command prompt x64 bit. Again, this is not a pybullet issue, use the pip support forums if they exist.

c3mcavus, 2020-12-31:
Thanks @erwincoumans. ```python setup.py develop``` worked! 

       ~ Download https://files.pythonhosted.org/packages/7b/4b/3dbd2ed72670b872e0bb26b83574dbf4f56e0d11a1a93a82d349ad4f6f31/pybullet-3.0.8.tar.gz
       ~ Extract the file ```...\Python37\Lib\pybullet-3.0.8```
       ~ Open terminal on this path
       ~ Finally, ```python setup.py developer```

ABHIINAV12, 2021-01-11:
> Never seen that error. You can also simply do
> 
> ```
> git clone https://github.com/bulletphysics/bullet3
> or download and unzip https://github.com/bulletphysics/bullet3/archive/master.zip and then:
> cd bullet3
> python setup.py build
> python setup.py install
> ```

hey, i am getting the error " Cannot open source file: 'examples/pybullet/pybullet.c': No such file or directory".
Can you help me out with this.

KhadijaHasnain, 2024-07-06:
Try using old version of Python 3.8 or less than try 
python -m pip install --upgrade pip
pip cache purge
pip install numpy
pip install pybullet --verbose
and it works