bulletphysics/bullet3 #1867: Improve the eglPlugin

erwincoumans, 2018-09-09:
Accepted the work-in-progress eglPlugin for rendering. Note that the plugin also works on Windows in DIRECT mode, using the Win32OpenGLWindow instead of EGLOpenGLWindow. I couldn't find a EGL implementation with OpenGL 3 for Windows / NVIDIA.

https://github.com/bulletphysics/bullet3/pull/1790

The conversion in the plugin to GLInstancingRendering is not fully implemented, track the progress in this issue.

erwincoumans, 2018-09-09:
Will move the eglPlugin from 'Extension' to static linked plugin in setup.py, it is easier to use that way.

BlGene, 2018-09-10:
A brief overview of related PRs:
1. #1597 is depreciated by merging of #1790
2. #1868 @dchichkov 's updates (superceded by #1869)
3.  #1869 @erwincoumans 's fixes (pose sync, build)

BlGene, 2018-09-10:
I've updated #1869 and fixed the `rendertest.py` and `rendertest_sync.py` scripts to do some profiling.

I noticed that rendering does not work if the `pybullet.loadPlugin` function is used outside a python context. see: https://github.com/BlGene/bullet3/blob/egl-fix/examples/pybullet/examples/rendertest.py#L132. I'm not sure why this fails, by maybe pybullet could raise a warning/error in this case.

`rendertest_sync.py` gives the following results on 

Desktop
```back nenv fps fps_tot
DIRECT/tiny 	 1 227 	 227
DIRECT/tiny 	 8 47 	 376
DIRECT/egl 	 1 115 	 115
DIRECT/egl 	 8 20 	 162
GUI/egl 	 1 294 	 294
GUI/egl 	 8 7 	 55
```

Server
```back nenv fps fps_tot
DIRECT/tiny 	 1 283 	 283
DIRECT/tiny 	 8 142 	 1133
DIRECT/egl 	 1 231 	 231
DIRECT/egl 	 8 77 	 613
GUI/egl 	 1 439 	 439
GUI/egl 	 8 68 	 545
```

If @dchichkov 's statement: 
> major performance gap between the plugin and GUI disappears once START_WIDTH / START_HEIGHT are hardcoded to correct values (although displayed GPU load is still much higher).

is still valid that might be a way to boost DIRECT/egl performance. 

Profiling looks as follows DIRECT/egl above and DIRECT/tiny below:
[image]

BlGene, 2018-09-12:
I looks like rendering sizes were not being correctly passed to the OpenGL renderer. This means that the the default resolution of 640x480 was being used, especially for small image sizes such as 84x84 this resulted in a 6x speed increase.

PR at #1875.

```
rendertest.py size: 84x84px
back nenv fps fps_tot
DIRECT/tiny 	 1 136 	 136
DIRECT/egl 	 1 992 	 992
GUI/egl 	 1 522 	 522

rendertest_sync.py size: 84x84px
back nenv fps fps_tot
DIRECT/tiny 	 1 126 	 126
DIRECT/tiny 	 8 47 	 380
DIRECT/egl 	 1 633 	 633
DIRECT/egl 	 8 167 	 1332
GUI/egl 	 1 419 	 419
GUI/egl 	 8 8 	 61

```

BlGene, 2018-10-03:
@erwincoumans 
If one loads a SDF file instead of a URDF file with the `testrender_egl.py` script the material colors don't work, with the default tiny renderer they do.

For example by substituting the R2D2 with the IIWA:
```
-pybullet.loadURDF("r2d2.urdf")
+pybullet.loadSDF("kuka_iiwa/kuka_with_gripper2.sdf")
```

BlGene, 2018-10-27:
@erwincoumans 
While the rendering stuff is still fresh, I think this is one of the few remaining bugs: Calling `pybullet.removeBody` does not seem to remove the object from the EGL renderer, it does so for the tiny renderer. Test script and images below.

https://gist.github.com/BlGene/938155a2c94b4e8dc7b88b7cbc037fa7

[image]

Also note: #1960 <- small pip-egl build fix

erwincoumans, 2018-10-29:
>>  Calling pybullet.removeBody does not seem to remove the object from the EGL renderer, 

Fixed here, thanks for the report: https://github.com/bulletphysics/bullet3/pull/1962
Note that you will run out of shape memory eventually, since we don't free the vertex/index data. So a 'resetSimulation' is needed after a while.

closed 2018-10-29 by erwincoumans

RchalYang, 2021-03-04:
Hi. It seems I'm still getting the same problem that pybullet.removeBody did not release GPU Memory with EGL.
I did use resetSimulation from time to time. I was able to reproduce the problem with example code and a few line more.

I'm using pybullet 3.0.8 & CUDA 11 

```
import pybullet as p
import time
import pkgutil
egl = pkgutil.get_loader('eglRenderer')
import pybullet_data

p.connect(p.DIRECT)
p.setAdditionalSearchPath(pybullet_data.getDataPath())
plugin = p.loadPlugin(egl.get_filename(), "_eglRendererPlugin")
print("plugin=", plugin)

p.configureDebugVisualizer(p.COV_ENABLE_RENDERING, 0)
p.configureDebugVisualizer(p.COV_ENABLE_GUI, 0)

p.setGravity(0, 0, -10)
idp = p.loadURDF("plane.urdf", [0, 0, -1])
r2d2 = p.loadURDF("r2d2.urdf")

pixelWidth = 320
pixelHeight = 220
camTargetPos = [0, 0, 0]
camDistance = 4
pitch = -10.0
roll = 0
upAxisIndex = 2

while (p.isConnected()):
    p.removeBody(idp)
    p.removeBody(r2d2)

    p.resetSimulation()

    idp = p.loadURDF("plane.urdf", [0, 0, -1])
    r2d2 = p.loadURDF("r2d2.urdf")

    
    for yaw in range(0, 360, 10):
        start = time.time()
        p.stepSimulation()
        stop = time.time()
        print("stepSimulation %f" % (stop - start))

        #viewMatrix = [1.0, 0.0, -0.0, 0.0, -0.0, 0.1736481785774231, -0.9848078489303589, 0.0, 0.0, 0.9848078489303589, 0.1736481785774231, 0.0, -0.0, -5.960464477539063e-08, -4.0, 1.0]
        viewMatrix = p.computeViewMatrixFromYawPitchRoll(camTargetPos, camDistance, yaw, pitch, roll,
                                                        upAxisIndex)
        projectionMatrix = [
            1.0825318098068237, 0.0, 0.0, 0.0, 0.0, 1.732050895690918, 0.0, 0.0, 0.0, 0.0,
            -1.0002000331878662, -1.0, 0.0, 0.0, -0.020002000033855438, 0.0
        ]

        start = time.time()
        img_arr = p.getCameraImage(pixelWidth,
                                pixelHeight,
                                viewMatrix=viewMatrix,
                                projectionMatrix=projectionMatrix,
                                shadow=1,
                                lightDirection=[1, 1, 1])
        stop = time.time()
        print("renderImage %f" % (stop - start))
        #time.sleep(.1)
        #print("img_arr=",img_arr)

p.unloadPlugin(plugin)
```