bulletphysics/bullet3 #1873: Unable to change the mass of link

ethanluoyc, 2018-09-11:
Hi,

I think there might be some issue with changing the mass of object
in the Gym environment. Specifically I am not able to change the mass of the link

```python
env = gym.make('HalfCheetahBulletEnv-v0')
unwrapped = env.unwrapped
unwrapped.reset()
p = unwrapped._p
part = unwrapped.robot.parts['torso']
bodyIndex = part.bodyIndex
partIndex = part.bodyPartIndex
info = p.getDynamicsInfo(part.bodies[bodyIndex], partIndex)
print('before mass', info[0])
p.changeDynamics(bodyIndex, partIndex, mass=5)
info = p.getDynamicsInfo(part.bodies[bodyIndex], partIndex)
print('after mass', info[0])
```

The output is 
```
WARN: gym.spaces.Box autodetected dtype as <class 'numpy.float32'>. Please provide explicit dtype.
WARN: gym.spaces.Box autodetected dtype as <class 'numpy.float32'>. Please provide explicit dtype.
WalkerBase::__init__ start
before mass 9.45733335583469
after mass 9.45733335583469
```
However this works for "InvertedPendulumBulletEnv-v0", if we change the mass of the pole (i.e., we
change the mass of the part "pole")

I am using Pybullet version 2.1.4 (just upgraded from PyPI)

erwincoumans, 2018-09-11:
You should never use the bodyPart 'bodyIndex' but use the bodyUniqueId, which is bodies[bodyIndex]

Attached is a modified script.

Note that the mass of the torso is intially 0, which is odd (it should compute the mass from the volume of the collision shapes of the link), there may be some issue in our simple MJCF parser.
[mass.zip](https://github.com/bulletphysics/bullet3/files/2371808/mass.zip)

closed 2018-09-11 by erwincoumans

ethanluoyc, 2018-09-12:
@erwincoumans Thanks for the clarification! Would it be good if these APIs and usage are documented somewhere?

ethanluoyc, 2018-09-12:
OK @erwincoumans, the reason you get 0 mass is because you are accessing partId=0, the torso has partId=3.

erwincoumans, 2018-09-12:
> @erwincoumans Thanks for the clarification! Would it be good if these APIs and usage are documented somewhere?

Have you seen the [PyBullet Quickstart Guide](https://docs.google.com/document/d/10sXEhzFRSnvFcl3XxNGhnD4N2SedqwdAvK3dsihxVUA/edit#heading=h.2ye70wns7io3)? It documents the 'changeDynamics' and other APIs.

ethanluoyc, 2018-09-12:
Yep. It's extremely well documented for the APIs, perhaps I should read the source code for pybullet_envs a little bit more closely to find out how I can grab the relevant parameters.