bulletphysics/bullet3 #1878: error: examples/OpenGLWindow/EGLOpenGLWindow.cpp: No such file or directory

pzhokhov, 2018-09-13:
Get this error when trying to install pybullet 2.1.6: 
```
gcc -pthread -B /opt/conda/compiler_compat -Wl,--sysroot=/ -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -Wstrict-prototypes -fPIC -I/opt/conda/lib/python3.6/site-packages/numpy/core/include -Iexamples/ThirdPartyLibs/optionalX11 -Isrc -Iexamples -Iexamples/ThirdPartyLibs -Iexamples/ThirdPartyLibs/glad -Iexamples/ThirdPartyLibs/enet/include -Iexamples/ThirdPartyLibs/clsocket/src -I/opt/conda/include/python3.6m -c examples/OpenGLWindow/EGLOpenGLWindow.cpp -o build/temp.linux-x86_64-3.6/examples/OpenGLWindow/EGLOpenGLWindow.o -DGWEN_COMPILE_STATIC -DBT_USE_DOUBLE_PRECISION -DBT_ENABLE_ENET -DBT_ENABLE_CLSOCKET -DB3_DUMP_PYTHON_VERSION -DPYBULLET_USE_NUMPY -D_LINUX -DGLEW_STATIC -DGLEW_INIT_OPENGL11_FUNCTIONS=1 -DGLEW_DYNAMIC_LOAD_ALL_GLX_FUNCTIONS=1 -DDYNAMIC_LOAD_X11_FUNCTIONS -DHAS_SOCKLEN_T -fno-inline-functions-called-once -DBT_USE_EGL -fPIC
    gcc: error: examples/OpenGLWindow/EGLOpenGLWindow.cpp: No such file or directory
```
no other errors in the error log (although a bunch of warnings)

erwincoumans, 2018-09-14:
Apparently some file was missing in the MANIFEST file, I reuploaded the a new 2.1.7 version to pypi,
can you try it again?
~~~~~~~~~~~~~~~~~~
pip install --no-cache-dir -U pybullet
~~~~~~~~~~~~~~~~~~

closed 2018-09-14 by erwincoumans

pzhokhov, 2018-09-14:
works, thanks!

mahaarbo, 2018-09-19:
I'm still getting this error:

```
x86_64-linux-gnu-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fno-strict-aliasing -Wdate-time -D_FORTIFY_SOURCE=2 -g -fstack-protector-strong -Wformat -Werror=format-security -fPIC -I/usr/lib/python2.7/dist-packages/numpy/core/include -Iexamples/ThirdPartyLibs/optionalX11 -Isrc -Iexamples -Iexamples/ThirdPartyLibs -Iexamples/ThirdPartyLibs/glad -Iexamples/ThirdPartyLibs/enet/include -Iexamples/ThirdPartyLibs/clsocket/src -I/usr/include/python2.7 -c examples/OpenGLWindow/EGLOpenGLWindow.cpp -o build/temp.linux-x86_64-2.7/examples/OpenGLWindow/EGLOpenGLWindow.o -DGWEN_COMPILE_STATIC -DBT_USE_DOUBLE_PRECISION -DBT_ENABLE_ENET -DBT_ENABLE_CLSOCKET -DB3_DUMP_PYTHON_VERSION -DPYBULLET_USE_NUMPY -D_LINUX -DGLEW_STATIC -DGLEW_INIT_OPENGL11_FUNCTIONS=1 -DGLEW_DYNAMIC_LOAD_ALL_GLX_FUNCTIONS=1 -DDYNAMIC_LOAD_X11_FUNCTIONS -DHAS_SOCKLEN_T -fno-inline-functions-called-once -DBT_USE_EGL -fPIC
    x86_64-linux-gnu-gcc: error: examples/OpenGLWindow/EGLOpenGLWindow.cpp: No such file or directory
    x86_64-linux-gnu-gcc: fatal error: no input files
    compilation terminated.
```

Running `pip search pybullet` gives `pybullet (2.1.8)`. This is on Ubuntu 16.04, running in parallels on a Mac OS X 10.11.6,

erwincoumans, 2018-09-19:
>> EGLOpenGLWindow.cpp: No such file or directory

should be fixed in 2.1.9, somehow the local fix didn't make it in pip. Thanks for the report

mahaarbo, 2018-09-19:
Excellent, keep up the good work!