bulletphysics/bullet3 #1881: problems with using .stl mesh

Anmol6, 2018-09-14:
Hi, 
I have a urdf [file](https://gist.github.com/Anmol6/6f3f69c089cb87b67bbc25b8457ab273) of a wheel and the mesh (which is basically a cylinder) is imported as a .stl [file](https://github.com/Anmol6/misc/blob/master/simplewheel.stl) (as done in the `testBike.py` example). However, when I load the wheel [here](https://gist.github.com/Anmol6/89c81a538678ea2fd5a35614593db377), it does not behave as a wheel; for eg. I let it drop from a height but when it hits the ground, it doesn't fully fall sideways.  

It could be that there's some problem in my .stl file, in which case it'd be great if you could recommend an easy-to-use software to create the .stl.  I used tinkerCad to create the .stl file.

erwincoumans, 2018-09-15:
The STL is not centered around the origin, so you need to compensate for it in the visual/collision transforms, try this one:

`<?xml version="0.0" ?>
<robot name="frame">
<link name="frontWheelLink">
    <inertial>
      <origin rpy="0 0 0" xyz="0 0 0"/>
       <mass value=".01"/>
       <inertia ixx="0.01" ixy="0" ixz="0" iyy="0.01" iyz="0" izz="0.01"/>
    </inertial>
    <visual>
      <origin rpy="0 0 0" xyz="-.2 0.3 0"/>
      <geometry>
      <mesh filename="simplewheel.stl" scale="0.001 0.001 0.001"/>
      </geometry>
        <material name="framemat0">
                <color rgba="0.9 0.4 0.3 1" />
        </material>
    </visual>
    <collision>
      <origin rpy="0 0 0" xyz="-.2 0.3 0"/>
      <geometry>
       <mesh filename="simplewheel.stl" scale="0.001 0.001 0.001"/>
      </geometry>
    </collision>
  </link>

</robot>`

closed 2018-09-15 by erwincoumans

erwincoumans, 2018-09-15:
By the way, I noticed your nice [simbicon pybullet project](https://github.com/Anmol6/locomotion). The upcoming version of Pybullet will make it easier to make a video, if ffmpeg is installed (Windows, Linux, Mac OSX): Attached is video and modified simbicon.py file.

[locomotion.zip](https://github.com/bulletphysics/bullet3/files/2385225/locomotion.zip)

Anmol6, 2018-09-15:
Thank you for the modified file as well as the Simbicon video! Yes, video recording would be great :) 

How did you know the STL's offset though?  For my purposes, I need to be as precise about this as possible.

dingzeyuli, 2019-05-08:
@Anmol6 Were you able to get the exact offset? 
@erwincoumans  I am using the the C++ API of btCompoundShape to import meshes. Do you know what would be the equivalent calls in C++ to set the offset? How is the offset computed? Is it the center of the bounding box? Thank you.

Anmol6, 2019-05-08:
@dingzeyuli Nope, I think tinkerCad doesn't provide nice offsets