bulletphysics/bullet3 #1888: Installation failed with exit status 1120

Count-Badgerston, 2018-09-20:
I have installed python 3.6 with Anaconda distribution, updated setuptools to 40.4.1 and pip to 18.0, then I installed visual studio 2017.

I also created a separate conda environment to install pybullet into.

Whenever I try to install pybullet with `pip install -U pybullet` I get error 
`command 'C:\\Program Files (x86)\\Microsoft Visual Studio\\2017\\Community\\VC\\Tools\\MSVC\\14.15.26726\\bin\\HostX86\\x64\\link.exe' failed with exit status 1120`. 

Is this a bug or am I doing something wrong ?

erwincoumans, 2018-09-20:
The essential part is missing, what file is missing from the linker. 
Can you copy and past the full outpu?
What version of pybullet (2.1.9 is latest)?

Second way:
Can you git checkout github.com/bulletphysics/bullet3 and run python setup.py build in the root of Bullet and provide full error message?

Count-Badgerston, 2018-09-20:
Alright, I gited the repository and ran `python setup.py install` and it did install !
The output was:
`Installed c:\anaconda3\envs\pybullet_env\lib\site-packages\pybullet-2.1.6-py3.6-win-amd64.egg
Processing dependencies for pybullet==2.1.6
Finished processing dependencies for pybullet==2.1.6`

It is not the latest version as I understand ?

Right after installation I tried to upgrade pybullet by running `pip install -U pybullet`. Not only it didn't upgrade to a newer version it also uninstalled the exiting one !

Then I ran `pip install -U pybullet ` again a recorded the output in a txt file. 
[output-log.txt](https://github.com/bulletphysics/bullet3/files/2402351/output-log.txt)

erwincoumans, 2018-09-20:
It is a ‘feature’ of pip to cache out-of-date versions. Use -no-cache-dir option

Count-Badgerston, 2018-09-21:
Probably it's just a problem with visual studio or some other component, not with pybullet itself. I haven't seen any other people posting the same issue.

erwincoumans, 2018-09-21:
We used to upload prebuild wheels, but there are too many incompatible Python components (regular versus Anaconda, with or without numpy, etc).

jvasqueza, 2020-04-12:
I am also trying to install pybullet in Windows10/Anaconda and I always get error>  ERROR: Command errored out with exit status 1: 'C:\anaconda\envs\myenv\python.exe' -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'C:\\Users\\jvasq\\AppData\\Local\\Temp\\pip-install-z6ar3h2u\\pybullet\\setup.py'"'"'; __file__='"'"'C:\\Users\\jvasq\\AppData\\Local\\Temp\\pip-install-z6ar3h2u\\pybullet\\setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record 'C:\Users\jvasq\AppData\Local\Temp\pip-record-3r7waoe5\install-record.txt' --single-version-externally-managed --compile --install-headers 'C:\anaconda\envs\myenv\Include\pybullet' Check the logs for full command output.