bulletphysics/bullet3 #1893: Pybullet install versions >2.1.6 fail with High Sierra

tomsilver, 2018-09-21:
I am running a new machine with High Sierra. My python version is 2.7.8, managed through pyenv.

I am unable to `pip install -U pybullet` (see errors below). I went back and tried `pip install pybullet==2.1.6` and was able to successfully install that version. I also had no problem with versions 2.1.4 and 2.1.3. However, versions 2.1.7 and 2.1.9 give what look like the same compile error.

Maybe related to https://github.com/bulletphysics/bullet3/issues/1892 and/or https://github.com/bulletphysics/bullet3/issues/1593

I am fine with 2.1.6 for now but wanted to report the issue. Thanks for your help!

The build fails:
```
  clang: error: no such file or directory: 'examples/OpenGLWindow/MacOpenGLWindowObjC.m'
  clang: error: no input files
  13 warnings generated.
  error: command 'clang' failed with exit status 1
  
  ----------------------------------------
  Failed building wheel for pybullet
  Running setup.py clean for pybullet
Failed to build pybullet
```

and then the final error is:
```
Command "/Users/tom/.pyenv/versions/2.7.8/bin/python2.7 -u -c "import setuptools, tokenize;__file__='/private/var/folders/2y/1l8njzj51qd7b7jnwzz5v2bm0000gn/T/pip-install-pabZ6T/pybullet/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /private/var/folders/2y/1l8njzj51qd7b7jnwzz5v2bm0000gn/T/pip-record-o6XXCI/install-record.txt --single-version-externally-managed --compile" failed with error code 1 in /private/var/folders/2y/1l8njzj51qd7b7jnwzz5v2bm0000gn/T/pip-install-pabZ6T/pybullet/
```

erwincoumans, 2018-09-21:
It should be fixed in 2.2.0, can you try it?
~~~~~~~
pip install --user -U  --no-cache-dir pybullet
~~~~~~~
Thanks for the report!

tomsilver, 2018-09-21:
Yes fixed! Thanks for the very fast turnaround.

closed 2018-09-21 by tomsilver

erwincoumans, 2018-09-21:
Sure, it bothers me as much as you if things are broken :-)

AndyLizd, 2019-03-26:
I have the following error:

error: [WinError 3] The system cannot find the path specified: 'C:\\Program Files (x86)\\Microsoft Visual Studio 14.0\\VC\\PlatformSDK\\lib'

Could anyone help?