bulletphysics/bullet3 #1904: 0 mean motor commands results in significant drift over time

Olimoyo, 2018-09-28:
Hi,

I am trying to reproduce the 2D reacher experiment using pybullet. As a quick test, I am sending random 0 mean actions:
```bash
        agent.applyAction(action=np.array([0, np.random.uniform(-1,1), np.random.uniform(-1,1),0,0,0]), type="velocity")
```
Over time it seems that the position of the joints drift away from 0. As shown by the pictures below:

Initial position:
[image]

Position after a while:
[image]

I made sure to turn off gravity and I'm sending velocity commands.

Does anyone have any idea why this might be happening or how to prevent this?

Thanks!

erwincoumans, 2018-09-28:
It is not clear what the issue is. When you apply random forces, forward dynamics will result in random poses. Why would the joint positions stay at 0, when applying random forces?

Olimoyo, 2018-09-28:
I might be misunderstanding what you are saying and/or what's going on in the underlying simulation but to clarify:

I am sending 0 mean random velocities at each time step. I thought that since the timesteps were constant and the velocities were 0 mean, the final position should remain near 0? My reasoning might be flawed, but I can't immedietaly see why. 

I don't think I disabled the velocity motor, I am using:
```bash
p.setJointMotorControlArray(id, motor_indices[0:6], p.VELOCITY_CONTROL,
                                            targetVelocities=v, forces=f)
```

erwincoumans, 2018-09-28:
~~~~~~~~~~~~~~
As a quick test, I am sending random 0 mean actions:
I thought that since the timesteps were constant and the velocities were 0 mean, the final position should remain near 0?
~~~~~~~~~~~~~~
Why would that be? Rotations are not commutative, so even if you step the simulation with a series of joint velocities whose sum is zero, the result on the position is surely not zero.

Try it with a cubic cube: apply a sequence of rotations, then apply the inverse sequence, but in randomized order. You surely don't end up in the same state.

Also, what does the test have to do with the environment? Just a basic sanity test?

Olimoyo, 2018-09-28:
Forgot to clarify this earlier, I'm also restricting the two joints to be planar relative to each other so that i'm dealing with 2D commutative rotations and the position is restricted to be in a plane. 

I did a quick test again, where I kept track of my sent average velocity commands and it seems like the position offset was just a function of how far off the actual average velocity was of zero.

Yup, it was just a basic sanity test since I am having a hard time learning the reacher-2D task in pybullet (bug is probably somewhere else). I was also initially concerned because in the original pictures the offset was especially bad, but after running it a few more times that situation seems to rarely happen (I am guessing since sampling np.random.uniform(-1,1) rarely goes too far off of 0).

erwincoumans, 2018-09-28:
Ah, so all is well on the sanity test?
~~~~~~~~~~~~
Yup, it was just a basic sanity test since I am having a hard time learning the reacher-2D task in pybullet (bug is probably somewhere else).
~~~~~~~~~~~~
Is this the 'ReacherBulletEnv-v0' PyBullet gym environment from pybullet_envs (part of pip install pybullet) or your custom env? What kind of bug?

Olimoyo, 2018-09-28:
Yup, I don't think the bug is related to pybullet itself and it was just a misunderstanding on my part.

It's my own custom environment but strongly based on 'ReacherBulletEnv-v0'. It just doesn't seem to get better as it trains (using OpenAI's baseline agent). I'll need to look into it more before specifying anything.

Thanks for the quick response and feedback Erwin!

closed 2018-09-28 by Olimoyo