bulletphysics/bullet3 #1908: how does pybullet compute lateral friction of spheres?

aznshodan, 2018-10-01:
I have two spheres. One has (mass, radius) of (2.0kg, 0.1m), and the other is (1.0kg, 0.1m).
It was obvious to me that when calling p.resetBaseVelocity to some velocity for both objects, the heavier one will travel shorter. However, after experimenting for a while, they travel the same distance.

So I ran p.getDynamicsInfo and turns out they both have the same lateral friction coefficient of 0.5. 
I can go and modify the urdf files of both spheres to some values but I want to know if those coefficients are in fact correct coefficients computed by pybullet internally. 

If not, do you know how I should compute the lateral coefficient, rolling coefficient, and spinning coefficient of sphere.urdf based on its mass and radius?

Thanks in advance.

underactuated, 2018-10-03:
I do not know how friction is implemented in pybullet, by usually the _friction coefficient_ does not depend on mass (do not confuse it with the _friction force_).

erwincoumans, 2018-12-22:
The friction coefficient is combined between two objects. If you want to know the lateral friction force for a certain contact point, see getContactPoints in the [PyBullet Quickstart Guide](https://docs.google.com/document/d/10sXEhzFRSnvFcl3XxNGhnD4N2SedqwdAvK3dsihxVUA/edit#heading=h.cb0co8y2vuvc).

closed 2018-12-22 by erwincoumans

LFGfg, 2024-10-28:
> The friction coefficient is combined between two objects. If you want to know the lateral friction force for a certain contact point, see getContactPoints in the [PyBullet Quickstart Guide](https://docs.google.com/document/d/10sXEhzFRSnvFcl3XxNGhnD4N2SedqwdAvK3dsihxVUA/edit#heading=h.cb0co8y2vuvc).

How to understand lateralFrictionDir1 and lateralFrictionDir2, and how are they defined?