bulletphysics/bullet3 #1912: Collision detection failed in fast speed

zanias589, 2018-10-02:
Hello guys,

I make a ball game using dynamic rigid body, I shoot the ball into a wall (static rigid body). The problem is when i shoot the ball in high velocity, the collision detection will failed and the ball will penetrates the wall.
Please help me to solve this issue because i am beginner using this, thanks

Guptajakala, 2018-10-25:
@zanias589  Why did you close? Have you solved?

zanias589, 2018-11-13:
Sorry, i found solution before. But I found this case again with smaller rates of happening.
I doing this
- create a method to predict future position at dt
- raycasting from current into predicted position
- if true, current position set into P = raycast point - ball radius * displacement.normalize)

calculation on contact point (collision, etc) still calculated by library on simulationStep().

@BluePrintRandom yes, please.

erwincoumans, 2019-02-08:
Using a ray test or sphere cast (with smaller radius than ball) can help. Or use a smaller fixed time step, so that the ball moves less than its radius in 1 simulation step.

closed 2019-02-08 by erwincoumans