bulletphysics/bullet3 #1916: Is the joint damping setting working correctly in mujoco models?

underactuated, 2018-10-03:
Hi,

I have posted a number of questions on the pybullet forum over the past couple months, but got no replies so far. I thought I would try my luck here.

I have tried to set the joint damping parameter to various values from 1 to 100000, but observed no effect in simulations. I simply set it as default joint parameters inside a URDF model file:

`<default>
    <joint armature="1" damping="100000" limited="false"/>
  </default>`

Am I doing it correctly? Thank you.

erwincoumans, 2019-02-05:
It is not extracted from the MuJoCo file, but I'm now exposing a PyBullet API to adjust the joint damping manually: 
pybullet.changeDynamics(objUid, linkIndex, jointDamping=...)
https://github.com/bulletphysics/bullet3/pull/2098/commits/42369aa47d41969cfcd85fd4858b3bb2ded2591a

erwincoumans, 2019-02-08:
joint damping is in now:
https://github.com/bulletphysics/bullet3/commit/42369aa47d41969cfcd85fd4858b3bb2ded2591a

closed 2019-02-08 by erwincoumans