bulletphysics/bullet3 #1921: Cannot install pybullet on Mac OS Mojave

BolunDai0216, 2018-10-09:
from examples/SharedMemory/plugins/tinyRendererPlugin/TinyRendererVisualShapeConverter.cpp:16:
    /usr/local/Cellar/gcc/8.2.0/include/c++/8.2.0/cwchar:44:10: fatal error: wchar.h: No such file or directory
     #include <wchar.h>
              ^~~~~~~~~
    compilation terminated.
    
    ----------------------------------------
Command "/anaconda3/bin/python -u -c "import setuptools, tokenize;__file__='/private/var/folders/26/3vt_y1v15td9p0rlbynzv72m0000gn/T/pip-install-z4tmfeia/pybullet/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /private/var/folders/26/3vt_y1v15td9p0rlbynzv72m0000gn/T/pip-record-m8lqizxf/install-record.txt --single-version-externally-managed --compile" failed with error code 1 in /private/var/folders/26/3vt_y1v15td9p0rlbynzv72m0000gn/T/pip-install-z4tmfeia/pybullet/

This is some of the error that I encountered. @erwincoumans

erwincoumans, 2018-10-09:
There is no information about a Bullet file that fails to compile.

Did you try pip install pybullet?
Or python setup.py build (in the root of Bullet)?

closed 2018-10-09 by erwincoumans

covarsha, 2018-11-22:
This link has the full error when I do pip install pybullet in conda 4.4.11, Python 3.6.3. https://docs.google.com/document/d/1DlFLc9sz-SsJh1lD918Yg2WykNiFtMTKzsY_JGf9evc/edit?usp=sharing
I am working on MacOS Mojave as well

erwincoumans, 2018-11-22:
your c++ environment is not properly setup.
Try 
-------------
xcode-select --instal
-------------
or google for missing string on mac ox

covarsha, 2018-11-22:
xcode-select --install gives the following output -
xcode-select: error: command line tools are already installed, use "Software Update" to install updates

I am not sure what else I can try to make this work. Do you have any other ideas? Thanks in advance!

samuelrohr, 2018-11-27:
I am facing the same problem.
macOS Mojave
Python 3.6
Conda 4.4.10
Xcode 10.1

Already tried remove the xcode_select tools e reinstall.

covarsha, 2018-11-27:
I installed it in a virtual environment without conda. It worked

BolunDai0216, 2018-11-27:
I have resolved this issue, it may be that you are using the anaconda distribution of python, you can check what python you are using by typing

which python

into the terminal. Changing the python to the version that I downloaded from python.org works for me. Also maybe using pip3 install would help.

Please let me know if this works, I got really frustrated when I was installing it, so I REALLY want to help you out!

erwincoumans, 2018-11-27:
Thanks for helping out. This is surely not  PyBullet issue but a MacOS +Python/conda+clang issue. You couls try compile from source using git clone this repo, then run python setup.py build and python setup.py install --user from the Bullet source tree root.

a1g0rithm, 2018-11-29:
@BolunDai0216 How do you switch from Anaconda dist to the d/l'd one from python.org?

BolunDai0216, 2018-11-29:
@a1g0rithm I created a virtual environment，and I specified the python distribution there

erwincoumans, 2018-11-29:
I just installed latest conda with python 3.7 and pip install pybullet works without issue.
It is not clear why some Mac OSX machines don't have conda/python/clang/xcode setup properly.

FunkSoulNinja, 2018-12-06:
I was having the same problem using `CFLAGS='-stdlib=libc++' pip install pybullet` worked for me.

erwincoumans, 2019-01-28:
I have been digging into this issue for a while and added the -stdlib=libc++ in the compile and linked step for Mac OSX in cpp files. Please check out PyBullet 2.4.2 or newer.

octavio-sosa, 2021-02-09:
These instructions for setting up Xcode worked for me:
https://github.com/nodejs/node-gyp/issues/569#issuecomment-255589932

They are consistent with @erwincoumans suggestion to try:
sudo xcode-select -switch /Applications/Xcode.app/Contents/Developer

kdm004, 2022-01-21:
I'm currently running MacOS High Sierra on my 2011 Macbook Pro, and I am trying to run a program that uses pybullet. I get the following error:

b3Printf: b3Warning[examples/Importers/ImportURDFDemo/BulletUrdfImporter.cpp,126]:

b3Printf: expected a world element, continuing without it.

I've tried to troubleshoot and it seems that MacOS High Sierra is not supported.

So now I've just tried to install pybullet on a new Macbook Pro 2018 that has MacOS Big Sur. I have all python3 requirements downloaded.

I receive the same error as one of the users above:
https://docs.google.com/document/d/1DlFLc9sz-SsJh1lD918Yg2WykNiFtMTKzsY_JGf9evc/edit

**My question is this:** If I install MacOS Big Sur on my 2011 Macbook, will I encounter this same installation failure with pybullet?