bulletphysics/bullet3 #1925: Support multiple simulations in PyBullet

maximecb, 2018-10-09:
I'm working on an OpenAI Gym environment, and would really like to see support for running multiple independent simulations in parallel. I realize that there are ways to hack this, either by offsetting the positions of simulations in space, or by forking processes, but these are very inconvenient.

erwincoumans, 2018-10-09:
Each PyBullet DIRECT connection runs independent. You can use the pybullet_utils/bullet_client.py to make it easier to run multiple independent DIRECT simulations. Each of those can run in its own thread without problem, no need to fork or offset in a shared environment

Do you have some suggestions what you have in mind to run environments in parallel?

maximecb, 2018-10-09:
I would like to see some example code for how to create multiple connections and use them independently. The only documentation I've seen is the quickstart guide, and the examples in that all have a single connection, and they don't thread any kind of "connection id" argument through the calls, to specify which simulation would be affected by the calls.

erwincoumans, 2018-10-09:
Thanks for the feedback, good points. Various environments (*) already can run in multiple thread (since our PPO runs them in parallel). You don't pass the connection id, that is kept internally to pybullet_client.py.

Let me get back to this, provide some docs and simple examples.

(*) PyBullet envs such as MinitaurReactiveEnv-v0, AntBulletEnv-v0, AntBulletEnv-v0, AntBulletEnv-v0, HumanoidFlagrunBulletEnv-v0, HumanoidFlagrunHarderBulletEnv-v0, HalfCheetahBulletEnv-v0, Walker2DBulletEnv-v0, StrikerBulletEnv-v0, InvertedPendulumSwingupBulletEnv-v0, InvertedDoublePendulumBulletEnv-v0, InvertedPendulumBulletEnv-v0.

erwincoumans, 2018-10-09:
See this example:
https://github.com/bulletphysics/bullet3/tree/master/examples/pybullet/gym/pybullet_utils/examples

If you installed pybullet using pip install pybullet, you can also run it like this:
python3 -m pybullet_utils.examples.multipleScenes

If you look at the following code, p0, p1 and pgui can run in parallel, in separate threads. Note that you can only have 1 GUI thread, the others have to be DIRECT, using TinyRenderer.

~~~~~~~~~~
import pybullet_utils.bullet_client as bc
import pybullet
import pybullet_data

p0 = bc.BulletClient(connection_mode=pybullet.DIRECT)
p0.setAdditionalSearchPath(pybullet_data.getDataPath())

p1 = bc.BulletClient(connection_mode=pybullet.DIRECT)
p1.setAdditionalSearchPath(pybullet_data.getDataPath())

#can also connect using different modes, GUI, SHARED_MEMORY, TCP, UDP, SHARED_MEMORY_SERVER, GUI_SERVER
pgui = bc.BulletClient(connection_mode=pybullet.GUI)

p0.loadURDF("r2d2.urdf")
p1.loadSDF("stadium.sdf")
print(p0._client)
print(p1._client)
print("p0.getNumBodies()=",p0.getNumBodies())
print("p1.getNumBodies()=",p1.getNumBodies())
while (1):
        p0.stepSimulation()
        p1.stepSimulation()
        pgui.stepSimulation()
~~~~~~~~~~

The default PyBullet environments use this bullet_client, instead of directly talking to pybullet, all API calls are automatically passed through with the right physicsClientId.

Is that what you wanted?

maximecb, 2018-10-09:
Yes, that is what I wanted. Thanks for responding with detailed information 👍

closed 2018-10-09 by erwincoumans