bulletphysics/bullet3 #1926: Support for procedural geometry in PyBullet

maximecb, 2018-10-09:
It seems like the only way to load a mesh is to read an obj file in. This is inconvenient for procedurally-generated geometry, which first has to be written to a file and then loaded again. It would be very convenient, and possibly faster, if we could directly pass an array of polygon/face data instead.

erwincoumans, 2018-10-26:
Attached is a patch, it just allows to pass vertices to createCollisionShape, to allow creation of convex hulls. Addition support for concave shapes requires indices. Adding support for visual shapes with textures requires uv coordinates.

I'm considering to upload data in-memory with in-memory file names instead of this patch. That way, we don't need to replicate all features of obj files. Do you require concave shapes and uv coords, normals etc?
[createColShapeVertices.zip](https://github.com/bulletphysics/bullet3/files/2517498/createColShapeVertices.zip)

~~~~~~~~~~~~~~~~~~~~
verts=[[-0.246350, -0.246483, -0.000624],
	[ -0.151407, -0.176325, 0.172867],
	[ -0.246350, 0.249205, -0.000624],
	[ -0.151407, 0.129477, 0.172867],
	[ 0.249338, -0.246483, -0.000624],
	[ 0.154395, -0.176325, 0.172867],
	[ 0.249338, 0.249205, -0.000624],
	[ 0.154395, 0.129477, 0.172867]]
stoneId = p.createCollisionShape(p.GEOM_MESH,vertices=verts)
~~~~~~~~~~~~~~~~~~~

maximecb, 2018-10-26:
Concave shapes yes, because the geometry I would create would be for rooms & hallways. UV coords no because I have my own rendering code. Normals, only if you need them for physics simulation.

erwincoumans, 2018-10-26:
OK, here is the PR to createCollisionShape for convex and concave collision shapes.
https://github.com/bulletphysics/bullet3/pull/1955
At the moment a limit of 1024 vertices, 1024 indices, will be lifted once streaming is implemented (soon).
Curious, what renderer do you use, any images/features/details?

maximecb, 2018-10-26:
I'm writing my own thing in Python/Pyglet. It's nothing impressive, but I'd rather be able to customize it in any way I want and not be tied to PyBullet.

Why do you require vertices+indices for concave? Do you not compute normals based on vertex winding?

erwincoumans, 2018-10-30:
https://github.com/bulletphysics/bullet3/pull/1955 was merged, so let's close the issue. If you need more than 1024 vertices or 1024 indices, please create a new issue.

closed 2018-10-30 by erwincoumans