bulletphysics/bullet3 #1928: pip install failure

BolunDai0216, 2018-10-10:
Last login: Wed Oct 10 00:16:37 on ttys000
Boluns-MacBook:~ bolundai$ pip install --user -U  --no-cache-dir pybullet
Collecting pybullet
  Downloading https://files.pythonhosted.org/packages/47/31/ffc261d1136595925987d00a9abe31dd11911f82e75a9386a8dedabbd4b4/pybullet-2.2.9.tar.gz (27.6MB)
    100% |████████████████████████████████| 27.6MB 15.6MB/s 
Installing collected packages: pybullet
  Running setup.py install for pybullet ... error
    Complete output from command /Library/Frameworks/Python.framework/Versions/3.6/bin/python3.6 -u -c "import setuptools, tokenize;__file__='/private/var/folders/37/hj3z6sw52q3dt_dy7ych14sc0000gn/T/pip-install-2c13_cih/pybullet/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /private/var/folders/37/hj3z6sw52q3dt_dy7ych14sc0000gn/T/pip-record-tim4_4np/install-record.txt --single-version-externally-managed --compile --user --prefix=:
    macosx-10.6-intel
    numpy is enabled.
    numpy_include_dirs = /Users/bolundai/Library/Python/3.6/lib/python/site-packages/numpy/core/include
    darwin!
    found resource files: 3232
    -- duckCM.png
    ...
    -- random_urdfs/640/640.mtl
    -- random_urdfs/640/640.obj
    -- random_urdfs/640/640.urdf
    ...
    packages
    ['pybullet_envs', 'pybullet_data', 'pybullet_utils', 'pybullet_envs.minitaur', 'pybullet_envs.prediction', 'pybullet_envs.agents', 'pybullet_envs.examples', 'pybullet_envs.bullet', 'pybullet_envs.baselines', 'pybullet_envs.minitaur.agents', 'pybullet_envs.minitaur.envs', 'pybullet_envs.minitaur.agents.tools', 'pybullet_envs.minitaur.agents.scripts', 'pybullet_envs.minitaur.agents.ppo', 'pybullet_envs.minitaur.envs.env_randomizers', 'pybullet_envs.agents.tools', 'pybullet_envs.agents.ppo', 'pybullet_data.policies', 'pybullet_data.policies.ppo', 'pybullet_data.policies.ppo.minitaur_reactive_env', 'pybullet_data.policies.ppo.minitaur_trotting_env', 'pybullet_utils.examples']
    -----
    running install
    running build
    running build_py
    creating build
    creating build/lib.macosx-10.6-intel-3.6
    creating build/lib.macosx-10.6-intel-3.6/pybullet_envs
    copying examples/pybullet/gym/pybullet_envs/env_bases.py -> build/lib.macosx-10.6-intel-3.6/pybullet_envs
    copying examples/pybullet/gym/pybullet_envs/gym_manipulator_envs.py -> build/lib.macosx-10.6-intel-3.6/pybullet_envs
    copying examples/pybullet/gym/pybullet_envs/gym_pendulum_envs.py -> build/lib.macosx-10.6-intel-3.6/pybullet_envs
    copying examples/pybullet/gym/pybullet_envs/robot_locomotors.py -> build/lib.macosx-10.6-intel-3.6/pybullet_envs
    copying examples/pybullet/gym/pybullet_envs/__init__.py -> build/lib.macosx-10.6-intel-3.6/pybullet_envs
    copying examples/pybullet/gym/pybullet_envs/robot_bases.py -> build/lib.macosx-10.6-intel-3.6/pybullet_envs
    copying examples/pybullet/gym/pybullet_envs/kerasrl_utils.py -> build/lib.macosx-10.6-intel-3.6/pybullet_envs
    copying examples/pybullet/gym/pybullet_envs/robot_pendula.py -> build/lib.macosx-10.6-intel-3.6/pybullet_envs
    copying examples/pybullet/gym/pybullet_envs/gym_locomotion_envs.py -> build/lib.macosx-10.6-intel-3.6/pybullet_envs
    copying examples/pybullet/gym/pybullet_envs/scene_stadium.py -> build/lib.macosx-10.6-intel-3.6/pybullet_envs
    copying examples/pybullet/gym/pybullet_envs/scene_abstract.py -> build/lib.macosx-10.6-intel-3.6/pybullet_envs
    copying examples/pybullet/gym/pybullet_envs/robot_manipulators.py -> build/lib.macosx-10.6-intel-3.6/pybullet_envs
    creating build/lib.macosx-10.6-intel-3.6/pybullet_data
    copying examples/pybullet/gym/pybullet_data/__init__.py -> build/lib.macosx-10.6-intel-3.6/pybullet_data
    creating build/lib.macosx-10.6-intel-3.6/pybullet_utils
    copying examples/pybullet/gym/pybullet_utils/bullet_client.py -> build/lib.macosx-10.6-intel-3.6/pybullet_utils
    copying examples/pybullet/gym/pybullet_utils/__init__.py -> build/lib.macosx-10.6-intel-3.6/pybullet_utils
    copying examples/pybullet/gym/pybullet_utils/urdfEditor.py -> build/lib.macosx-10.6-intel-3.6/pybullet_utils
    creating build/lib.macosx-10.6-intel-3.6/pybullet_envs/minitaur
    copying examples/pybullet/gym/pybullet_envs/minitaur/__init__.py -> build/lib.macosx-10.6-intel-3.6/pybullet_envs/minitaur
    creating build/lib.macosx-10.6-intel-3.6/pybullet_envs/prediction
    copying examples/pybullet/gym/pybullet_envs/prediction/boxstack_pybullet_sim.py -> build/lib.macosx-10.6-intel-3.6/pybullet_envs/prediction
    copying examples/pybullet/gym/pybullet_envs/prediction/__init__.py -> build/lib.macosx-10.6-intel-3.6/pybullet_envs/prediction
    copying examples/pybullet/gym/pybullet_envs/prediction/test_pybullet_sim_gym_env.py -> build/lib.macosx-10.6-intel-3.6/pybullet_envs/prediction
    copying examples/pybullet/gym/pybullet_envs/prediction/pybullet_sim_gym_env.py -> build/lib.macosx-10.6-intel-3.6/pybullet_envs/prediction
    creating build/lib.macosx-10.6-intel-3.6/pybullet_envs/agents
    copying examples/pybullet/gym/pybullet_envs/agents/configs.py -> build/lib.macosx-10.6-intel-3.6/pybullet_envs/agents
    copying examples/pybullet/gym/pybullet_envs/agents/train_ppo.py -> build/lib.macosx-10.6-intel-3.6/pybullet_envs/agents
    copying examples/pybullet/gym/pybullet_envs/agents/utility.py -> build/lib.macosx-10.6-intel-3.6/pybullet_envs/agents
    copying examples/pybullet/gym/pybullet_envs/agents/visualize_ppo.py -> build/lib.macosx-10.6-intel-3.6/pybullet_envs/agents
    copying examples/pybullet/gym/pybullet_envs/agents/__init__.py -> build/lib.macosx-10.6-intel-3.6/pybullet_envs/agents
    copying examples/pybullet/gym/pybullet_envs/agents/networks.py -> build/lib.macosx-10.6-intel-3.6/pybullet_envs/agents
    creating build/lib.macosx-10.6-intel-3.6/pybullet_envs/examples
    ...
    warning: include path for stdlibc++ headers not found; pass '-std=libc++' on the command line to use the libc++ standard library instead [-Wstdlibcxx-not-found]
    warning: include path for stdlibc++ headers not found; pass '-std=libc++' on the command line to use the libc++ standard library instead [-Wstdlibcxx-not-found]
    examples/TinyRenderer/our_gl.cpp:1:10: fatal error: 'cmath' file not found
    In file included from examples/TinyRenderer/TinyRenderer.cpp:1:
    In file included from examples/TinyRenderer/TinyRenderer.h:4:
    examples/TinyRenderer/geometry.h:3:10: fatal error: 'cmath' file not found
    #include <cmath>
             ^~~~~~~
[... 1007 lines cut ...]
    1 warning and 1 error generated.
    1 warning generated.
    warning: include path for stdlibc++ headers not found; pass '-std=libc++' on the command line to use the libc++ standard library instead [-Wstdlibcxx-not-found]
    In file included from examples/OpenGLWindow/MacOpenGLWindowObjC.m:7:
    In file included from examples/OpenGLWindow/OpenGLInclude.h:28:
    examples/ThirdPartyLibs/glad/glad/gl.h:2010:19: error: typedef redefinition with different types ('ptrdiff_t' (aka 'int') vs 'intptr_t' (aka 'long'))
    typedef ptrdiff_t GLintptrARB;
                      ^
    /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/System/Library/Frameworks/OpenGL.framework/Headers/gltypes.h:55:18: note: previous definition is here
    typedef intptr_t GLintptrARB;
                     ^
    In file included from examples/OpenGLWindow/MacOpenGLWindowObjC.m:7:
    In file included from examples/OpenGLWindow/OpenGLInclude.h:28:
    examples/ThirdPartyLibs/glad/glad/gl.h:2015:19: error: typedef redefinition with different types ('ptrdiff_t' (aka 'int') vs 'intptr_t' (aka 'long'))
    typedef ptrdiff_t GLsizeiptrARB;
                      ^
    /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/System/Library/Frameworks/OpenGL.framework/Headers/gltypes.h:56:18: note: previous definition is here
    typedef intptr_t GLsizeiptrARB;
                     ^
    2 warnings and 2 errors generated.
    1 warning generated.
    
    ----------------------------------------
Command "/Library/Frameworks/Python.framework/Versions/3.6/bin/python3.6 -u -c "import setuptools, tokenize;__file__='/private/var/folders/37/hj3z6sw52q3dt_dy7ych14sc0000gn/T/pip-install-2c13_cih/pybullet/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /private/var/folders/37/hj3z6sw52q3dt_dy7ych14sc0000gn/T/pip-record-tim4_4np/install-record.txt --single-version-externally-managed --compile --user --prefix=" failed with error code 1 in /private/var/folders/37/hj3z6sw52q3dt_dy7ych14sc0000gn/T/pip-install-2c13_cih/pybullet/
Boluns-MacBook:~ bolundai$

BolunDai0216, 2018-10-10:
Due to the upload limit I tried to upload as much as I can, this is what happened during my installation. I really don't know what this means please help me take a look and see what the issue is, thanks!

erwincoumans, 2018-10-10:
please attach full log in a zip file

erwincoumans, 2018-10-10:
Your system, xcode and command-line tools is not properly setup. Google for 'fatal error: 'algorithm' file not found'

closed 2018-10-10 by erwincoumans

BolunDai0216, 2018-10-10:
I attached the log to this comment
[Terminal Saved Output.txt](https://github.com/bulletphysics/bullet3/files/2465476/Terminal.Saved.Output.txt)

erwincoumans, 2018-10-10:
Your system, xcode and command-line tools is not properly setup. Google for 'fatal error: 'algorithm' file not found'.

We don't have the bandwidth to help with that, please search on the internet for a solution on the errors.

BolunDai0216, 2018-10-15:
I found what the problem is, I was trying to install it will the anaconda version of python 3.6.5 and failed, however when I was trying to install it for python 3.7 it can be successfully installed, however for 3.7 it doesn't support tensorflow. Can you specify what version of python you are using?