bulletphysics/bullet3 #1932: What's the meaning of depthPixels?

JobSmith, 2018-10-13:
The getCameraImage function returns the depthPixels by inputing viewMatrix and projectionMatrix.
I want to know the method of this calculating process and the meaning of the depthPixels.
I find its values ranges from zero to one.
What is the measuring scale or rate of it?

erwincoumans, 2018-10-13:
Bullet uses OpenGL to render, and the convention is non-linear z-buffer.

See https://stackoverflow.com/questions/6652253/getting-the-true-z-value-from-the-depth-buffer

far=1000.//depends on projection matrix, this is default
near=0.01//depends on projection matrix
depth = far * near / (far - (far - near) * depthImg)//depthImg is the depth from Bullet 'getCameraImage'

Note that this 'true' z value is still not really the distance from 3d point to the camera center,
but the projected distance onto the near plane.
If you want the actual distance from camera to 3d point, you need a bit more work.
depth/=math.cos(alpha)

See PyBullet https://github.com/bulletphysics/bullet3/blob/master/examples/pybullet/examples/pointCloudFromCameraImage.py

closed 2018-10-13 by erwincoumans

erwincoumans, 2018-10-17:
It means your PyBullet is compiled without NumPy and return flat Python list. Make sure your Python has Numpy and install usiny pip install pybullet.

JobSmith, 2018-10-18:
In fact, I have already installed numpy.
[image]