bulletphysics/bullet3 #1933: Issue loading meshes from URDF

davidgolub, 2018-10-13:
When loading a simple mesh from the following URDF, all of the color is gone without warning, even though the texture is there. But when I load the same mesh using p.createVisualShape(...), p.createMultibody(...) it loads fine. Any thoughts on what could be happening?  Thanks!

![image](
[image]
)

``

erwincoumans, 2018-10-13:
Only 1 mesh/color per link.
You can change the color using
p.changeVisualShape(obUniqueId, linkIndex, rgbaColor=[r,g,b,a])
Otherwise, share the files, zipped.

davidgolub, 2018-10-13:
Yes, there is only 1 mesh per a link. Will share the files right now.

davidgolub, 2018-10-13:
[globe.tar.gz](https://github.com/bulletphysics/bullet3/files/2475846/globe.tar.gz)
The urdf file is in configuration_2017-07-18/globe.urdf, and the meshes are in the meshes folder. Note that you would have to change the absolute path of meshes in the urdf file to point to the proper meshes subdirectory on your machine

davidgolub, 2018-10-13:
Thanks for your fast response. But in the URDF file we already converted all objects from DAE to OBJ format, and we're still facing the same issue.

erwincoumans, 2018-10-13:
The color can be
A) assigned in the URDF file
~~~~~~~~~~~~
<link name ="rb1_marker">
    <visual>
      <origin xyz='-0.125441 0.088169 -0.129482' rpy='-1.936752 0.767602 1.007135'/>
      <geometry>
        <mesh filename="meshes/globe_sphere.dae"  scale="1 1 1" />
      </geometry>
       <material name="red">
        <color rgba="1 0 0 1"/>
      </material>
    </visual>
  </link>
~~~~~~~~~~~~
B) assigned in PyBullet
~~~~~~~~~~~
p.changeVisualShape(obUniqueId, linkIndex, rgbaColor=[r,g,b,a])
~~~~~~~~~~~

Both versions work when I test the zip file.
Did you try method A and/or B?

Which of those don't work?