bulletphysics/bullet3 #1936: Applying force to grasp an object so that the object does not slip through the gripper

jazzscout, 2018-10-15:
Hi,

I am working with pybullet to grasp various objects using a 6DOF robotic arm. I am wondering if there is any way through which we could mention that how much force should be applied on the gripper/fingers of gripper so that the object does not slide/slip through the hand while trying to pick it? 

As for now, I am using position control to grasp the objects but when the robot tries to pick them up, the objects fall from its gripper. So could anyone share their thoughts or any advice in this regard?

Best

charles-blouin, 2018-10-23:
You need to use force control. As for the force required, it depends on the orientation of the gripper and the object, the geometry, the acceleration of the path chosen, and the friction coefficient between the gripper and the object. It is a complex problem, there is no simple solution. Could you just try different holding force until it works for your application?

erwincoumans, 2018-10-30:
There are a few examples of a robot arm with gripper, using position control.

python3 -m pybullet_envs.baselines.enjoy_kuka_diverse_object_grasping
https://github.com/bulletphysics/bullet3/blob/master/examples/pybullet/gym/pybullet_envs/baselines/enjoy_kuka_diverse_object_grasping.py and
https://github.com/bulletphysics/bullet3/blob/master/examples/pybullet/gym/pybullet_envs/bullet/kuka.py

It all depends on the problem and settings. Large mass ratios can cause issues. Gripper geometry too (use boxes first instead of meshes). Timestep setting (smaller is better), number of solver iterations (use more and/or use solverResidualThreshold) etc. etc.
POSITION_CONTROL and VELOCITY_CONTROL requires a maxForce not too large.

Be careful with rolling/spinning friction (keep it 0 or very close to 0). Consider using friction_anchors (see urdf files) etc.

closed 2018-10-30 by erwincoumans