bulletphysics/bullet3 #1939: Installation of PyBullet 2.3.0 fails on Ubuntu 16.04

Kukanani, 2018-10-16:
I'm working in a fresh Python 3.5.2 virtualenv, and I'm unable to install pybullet 2.3.0 via `pip install pybullet`. 2.2.9 can still be installed successfully.

The terminal output is very large (over 20k lines!), so I'm attaching it as a file for ease of use:

[install_pybullet_230_output.txt](https://github.com/bulletphysics/bullet3/files/2483618/install_pybullet_230_output.txt)

erwincoumans, 2018-10-16:
Thanks for the report. Some file is missing, I'll update the MANIFEST.in and upload version 2.3.1.
Please try it again in a few minutes, using
pip install --no-cache-dir pybullet --user

closed 2018-10-16 by erwincoumans

Kukanani, 2018-10-16:
Verified working on my machine. Thanks!