bulletphysics/bullet3 #1947: rayTest() produces inaccurate results

dkogan, 2018-10-22:
Hi. I'm using rayTest() to compute ray-object intersections. This appears to have accuracy issues even with very simple geometry, and tweaking the various margins doesn't appear to change anything. It's very possible that I'm simply doing something wrong and this isn't a bug, but I'd appreciate a pointer.

I create one geometry object: a very wide wall in the xy plane, offset in the z direction. From the side I'm looking, the wall sits at z=50. So all reported wall-ray intersections should have z=50. I shoot rays at the wall that are mostly perpendicular; they're tilted down slightly. I translate the ray along the x axis, and for each position I ask Bullet to compute the intersection, and I record the result. This motion is along the wall. The results look like this:

http://notes.secretsauce.net/bullettest.pdf

So if x<0, I get the expected z=50, but for x>0 I don't, with less and less accuracy as I move away from x=0. With this test, things get as bad as ~49.6: 0.8% error. The wall runs from x=-500 to x=500, so we shouldn't be sensitive to the x. Suggestions? Is this expected? How much error should I expect? Should I be adjusting some margin parameter somewhere to get better results?

The test program lives here:

http://notes.secretsauce.net/bullettest.cc

I build it like this:

    g++ -o bullettest `pkg-config --cflags --libs bullet` bullettest.cc

This program takes an x value on the commandline. I scan the x coordinates like this:

    ( echo '# x0 x y z'; for x0 in `seq -250.0 0.1 250.0`; do ./bullettest $x0; done ) > bullettest.vnl

The resulting bullettest.vnl is here:

http://notes.secretsauce.net/bullettest.vnl

I then made the above plot like this

    < bullettest.vnl vnl-filter -p x0,z | feedgnuplot --points --domain --xlabel 'X offset of the start, end of the ray' --ylabel 'z coord of the intersection. Should be exactly 50' --hardcopy bullettest.pdf

This is on a recent Debian/sid box, with bullet packages version 2.87+dfsg-2. I've observed an identical behavior on a recent-ish Fedora release, so this isn't Debian-specific.

Thanks

dkogan, 2018-10-22:
I just ran another slightly-simpler test. Apparently it's not required to tilt the rays down to observe this problem. Shooting the rays from (x,0,0) to (x,0,100) is enough to see things work correctly for x<0 and incorrectly for x>0.

dkogan, 2018-10-24:
Poking around, I see that the inaccurate distances reported for x>0 come from exceeding MAX_ITERATIONS in btSubsimplexConvexCast::calcTimeOfImpact(). There's a comment in the source that mentions an internet discussion about this topic, but the link is dead. As an experiment, I bumped up the MAX_ITERATIONS by a factor of 10, and it doesn't help: we still hit the new iteration limits, and the accuracy does not improve.

I tried to bypass this code path entirely by selecting a different method:

    closestResults.m_flags |=
        btTriangleRaycastCallback::kF_UseGjkConvexCastRaytest;

This changes the distribution of errors (we generally overestimate the distance instead of underestimating it, and we do so for both x<0 and x>0). But the magnitude of errors is still high. I traced the sources. Here there's an iterative solver of some sort in btGjkConvexCast::calcTimeOfImpact(). I looked at the most inaccurate point in my set. Here we don't exceed the iteration count; it looks like we return normally, albeit with a wrong result.

Ideas?

erwincoumans, 2018-10-24:
Please try again using the latest Bullet github repo.
Are you using double precision or single precision? What compiler version / platform?
I tried running your test, but didn't see the approximations.
Using latest Bullet from github, in double precision: 

[image]

In single precision, fp:fast:

[image]

Obviously, we can (and should) implement the exact raytest for box and convex shapes, instead of using the iterative algorithm. Latest Bullet uses the GJK convex cast and has some improvements in GJK.

erwincoumans, 2018-10-24:
It uses a slightly modified version of yours, fixing a compile problem and iterating in C++ (instead of from args), and printf a comma-separated values (csv). Please use this version, for comparison:
[RayTest.zip](https://github.com/bulletphysics/bullet3/files/2512764/RayTest.zip)

dkogan, 2018-10-25:
Thanks for the replies. I'm using a mostly bleeding edge install of Debian/sid. GNU/Linux. amd64. This was built with single-precision floating point. To make debugging easier, I'm now running against a bleeding edge clone of your repo. I just cloned it, and I'm at 9c1db5fb349514df62e2a9ba44cc343f4c7c8018.

I'm also now using your test application with one tiny change: to ease debugging I'm now shooting the rays directly towards the wall, without any downward tilt. The relevant lines now look like this:

		btVector3 btFrom ((btScalar)x, 0, 0);
		btVector3 btTo   ((btScalar)x, 0, 100.0);

I have rebuilt bullet with single-precision floating point, and I am seeing the same exact issues as before. I.e. updating to the latest code didn't change anything.

I then rebuilt it with double-precision floating point:

     cmake -DUSE_DOUBLE_PRECISION=On

After doing that the test application crashes here:

		_broadphase = &_overlap_pair_cache;

Full backtrace:

```
(gdb) where
#0  btAlignedObjectArray<btAlignedObjectArray<btDbvtNode const*> >::resize (fillData=..., newsize=65536, this=0x7fffffff89b0) at /tmp/bullet3/src/LinearMath/btAlignedObjectArray.h:218
#1  btAlignedObjectArray<btAlignedObjectArray<btDbvtNode const*> >::copyFromArray (otherArray=..., this=0x7fffffff89b0) at /tmp/bullet3/src/LinearMath/btAlignedObjectArray.h:506
#2  btAlignedObjectArray<btAlignedObjectArray<btDbvtNode const*> >::operator= (other=..., this=0x7fffffff89b0) at /tmp/bullet3/src/LinearMath/btAlignedObjectArray.h:66
#3  btDbvtBroadphase::operator= (this=<optimized out>) at /tmp/bullet3/src/BulletCollision/BroadphaseCollision/btDbvtBroadphase.h:62
#4  main () at bullettest2.cc:11
#5  0x00007ffff7ab8b17 in __libc_start_main (main=0x55555555cda0 <main>, argc=1, argv=0x7fffffffdde8, init=<optimized out>, fini=<optimized out>, rtld_fini=<optimized out>, stack_end=0x7fffffffddd8) at ../csu/libc-start.c:310
#6  0x000055555555edaa in _start () at /usr/include/c++/8/new:169
```

I haven't yet done any more debugging than this. Should I dig into it, or do you know off the top of your head what the issue is? There're no stale binaries sitting around. I'm linking to bullet statically.

erwincoumans, 2018-10-25:
I don't have the bandwidth to do the debugging for you.
Please do a fresh checkout, and use premake4 to build stuff instead, and copy that file over examples/HelloWorld/HelloWorld.cpp, so you don't need to make ANY change in the build system etc.
~~~~~~~~
cd build3
./premake4_linux64 gmake
cd gmake
make App_HelloWorld
../../bin/App_HelloWorld_<whatever> 
~~~~~~~~

erwincoumans, 2018-10-25:
~~~~~~~~~
_broadphase = &_overlap_pair_cache;
~~~~~~~~~
This makes no sense, remote those lines and only use:
~~~~~~~~~
btDbvtBroadphase             _broadphase;
~~~~~~~~~

dkogan, 2018-10-25:
I made that correction, and it doesn't make any difference with the results I'm seeing.

As for the compiler, I don't know nearly enough about the bullet internals to say it's doing a "bad job". It's gcc 8.2.0. If there are issues with this compiler and bullet, then many people would be affected. I'm happy to help you debug, if you like.

In any case, this addresses my immediate problem, so I'm closing this issue. Thank you very much!

closed 2018-10-25 by dkogan

erwincoumans, 2018-10-25:
I just tried it on a linux box and also got the inaccurate result with single precision. I'll have a look at the single precision issue for large shape ratios at some stage. Thanks for the repro case!

Note that you also may want to checkout pybullet, which has a lot of interesting features. It compiles using double precision by default as well, since it targets robotics and reinforcement learning.