bulletphysics/bullet3 #1952: How to project 3D point onto image plane?

msieb1, 2018-10-24:
I am multiplying P*V*p3d where P and V are the 4x4-reshaped projection and view matrix, respectively. However, this does not yield the corresponding pixel location of the used 3D point when checking with image obtained with p.getCameraImage.

What is the correct way to compute the 2D projection by hand?

msieb1, 2018-10-25:
Thanks for the answer. I tried exactly that, but the xy values are out of the [-1,1] range after normalizing. We do not have to take into account the Model matrix, right?

daxum, 2018-10-25:
It depends on if the point you're using is in world space or model space. From what you've said, it seems that it is in world space, so that shouldn't be a problem. X and y being outside [-1, 1] means that the point's offscreen, so are you sure it's visible?

erwincoumans, 2018-10-28:
You may want to ask such question in an OpenGL forum or learning more about TinyRenderer, it has nice documentation.https://github.com/ssloy/tinyrenderer/wiki
(we use it for our software CPU rendering: 
https://github.com/bulletphysics/bullet3/blob/master/examples/TinyRenderer/our_gl.cpp#L82)

closed 2018-10-28 by erwincoumans

neha191091, 2020-02-02:
Hi @msieb1,
Were you able to find a solution to the issue you were having? I am currently facing a similar issue, where I am not getting the normalized pixel values to lie in [-1, 1] even when the corresponding 3d coordinate shows up in my image.