bulletphysics/bullet3 #1954: Pybullet collisioin not detected when reading STL in URDF

Guptajakala, 2018-10-25:
I'm switching to pybullet from gazebo after finding its friendly functions. However, when I'm using the same gripper urdf file and tested in pybullet and gazebo and tried to close the gripper, pybullet cannot detect the collision correctly. I flagged URDF_USE_SELF_COLLISION as:
p.loadURDF(filename, pos, quat, flags=p.URDF_USE_SELF_COLLISION)

In gazebo:
https://drive.google.com/open?id=1oGOGCzv11F7pgGmLpEmy4PfXaUFaMtu2

In pybullet:
https://drive.google.com/open?id=1K7qY0POYECLg6Dj42ll5EuGl2rNdbE24

My project:
https://drive.google.com/open?id=1xTBU731nyNXdtrTvGRZFg0UXreHyWK6g
(run main.py can reproduct the problem)

I really appreciate it if someone could explain so that I can continue using pybullet.

Guptajakala, 2018-10-25:
@erwincoumans Sorry for the previous, I updated a new one.

erwincoumans, 2018-10-27:
There are numerous issues with your project. Damping should be close to 0, never above 1. Rolling/spinning friction should be 0 or close to 0 (never >1). I removed all contact sections from the URDF and a few more changes. Now it works better. It is best to replace the too detailed finger mesh with blocks/cube primitives. Also, that block is 0.02 gram, which doesn't work well with the default number of iterations etc.
[src.zip](https://github.com/bulletphysics/bullet3/files/2521394/src.zip)
In a nutshell, you may want to stick with Gazebo if that works well for you, since we don't have a support team helping you.

closed 2018-10-27 by erwincoumans

Guptajakala, 2018-10-27:
@erwincoumans ,  Awesome, now it works much better! Also thank you for your very helpful suggestions, I will take care of those in the future. I think I can happily continue my project in Pybullet now :)