bulletphysics/bullet3 #1956: pybulllet installation stuck on setup.py

pipriles, 2018-10-26:
Hello,

I have python 3.7.1 and i have tried several times with:

`sudo pip install pybullet`
`sudo pip install --no-cache-dir pybullet`

And i just get to:
```
  Running setup.py install for PyBullet ... /
```

And it keeps there for a long time...
I have even downloaded the code and compiled it, and did `python setup.py` and nothing.

I have tried to install it on my laptop and it works, what am i missing?

Thanks.

erwincoumans, 2018-10-26:
Install should take less than a few minutes. What is the output of
~~~~~~~~~~~~~~
python setup.py build
~~~~~~~~~~~~~~

pipriles, 2018-10-27:
OS: Arch Linux
G++: 8.2.1

Output from `python setup.py build`

```
[...]
-- mjcf/humanoid_symmetric.xml
-- mjcf/ground_plane.xml
-- mjcf/humanoid.xml
-- mjcf/thrower.xml
-- mjcf/ground.xml
-- mjcf/humanoid_fixed.xml
-- mjcf/cylinder_fromtoY.xml
-- mjcf/ant.xml
-- mjcf/swimmer.xml
-- table/table.png
-- table/table.urdf
-- table/table.mtl
-- table/table.obj
packages
['pybullet_envs', 'pybullet_utils', 'pybullet_data', 'pybullet_envs.minitaur', 'pybullet_envs.prediction', 'pybullet_envs.examples', 'pybullet_envs.baselines', 'pybullet_envs.bullet', 'pybullet_envs.agents', 'pybullet_envs.minitaur.envs', 'pybullet_envs.minitaur.agents', 'pybullet_envs.minitaur.envs.env_randomizers', 'pybullet_envs.minitaur.agents.ppo', 'pybullet_envs.minitaur.agents.tools', 'pybullet_envs.minitaur.agents.scripts', 'pybullet_envs.agents.ppo', 'pybullet_envs.agents.tools', 'pybullet_utils.examples', 'pybullet_data.policies', 'pybullet_data.policies.ppo', 'pybullet_data.policies.ppo.minitaur_reactive_env', 'pybullet_data.policies.ppo.minitaur_trotting_env']
-----
running build
running build_py
running build_ext
building 'pybullet' extension

```

And doesn't continue... :(

erwincoumans, 2018-10-27:
Can you edit the setup.py and remove the monkey-patch?

~~~~~
# monkey-patch for parallel compilation
import multiprocessing
import multiprocessing.pool
def parallelCCompile(self, sources, output_dir=None, macros=None, include_dirs=None, debug=0, extra_preargs=None, extra_postargs=None, depends=None):
    # those lines are copied from distutils.ccompiler.CCompiler directly
    macros, objects, extra_postargs, pp_opts, build = self._setup_compile(output_dir, macros, include_dirs, sources, depends, extra_postargs)
    cc_args = self._get_cc_args(pp_opts, debug, extra_preargs)
    # parallel code
    N = 2*multiprocessing.cpu_count()# number of parallel compilations
    def _single_compile(obj):
        try: src, ext = build[obj]
        except KeyError: return
        self._compile(obj, src, ext, cc_args, extra_postargs, pp_opts)
    # convert to list, imap is evaluated on-demand
    list(multiprocessing.pool.ThreadPool(N).imap(_single_compile,objects))
    return objects
import distutils.ccompiler
distutils.ccompiler.CCompiler.compile=parallelCCompile
~~~~~

Or unzip and copy the attached setup.py and see if that works.

[setup.zip](https://github.com/bulletphysics/bullet3/files/2521334/setup.zip)

pipriles, 2018-10-27:
I have deleted it and now it works, thank you!

erwincoumans, 2018-10-27:
Thanks. Pity the monkey-patch doesn't work on all platforms. If there are many more reports like this, I'll disable it.

closed 2018-10-27 by erwincoumans

pipriles, 2018-10-27:
It's very weird because my laptop has the same distro and the same compiler...

erwincoumans, 2018-10-28:
Could you help debugging the issue? 
~~~~~~~
N = 2*multiprocessing.cpu_count()# number of parallel compilations
print("N=",N)
~~~~~~~
What is the return value of N on your machine? Or add some additional print statements to find where it stalls.

pipriles, 2018-11-02:
Hello!
I think it has to do something with python internally, because i have changed the line where you iterate the lazy imap from pool using a list constructor with a for and it worked...

```
# monkey-patch for parallel compilation
import multiprocessing
import multiprocessing.pool
def parallelCCompile(self, sources, output_dir=None, macros=None, include_dirs=None, debug=0, extra_preargs=None, extra_postargs=None, depends=None):
    # those lines are copied from distutils.ccompiler.CCompiler directly
    macros, objects, extra_postargs, pp_opts, build = self._setup_compile(output_dir, macros, include_dirs, sources, depends, extra_postargs)
    cc_args = self._get_cc_args(pp_opts, debug, extra_preargs)
    # parallel code
    N = 2*multiprocessing.cpu_count()# number of parallel compilations
    def _single_compile(obj):
        try: src, ext = build[obj]
        except KeyError: return
        self._compile(obj, src, ext, cc_args, extra_postargs, pp_opts)
    # convert to list, imap is evaluated on-demand
    pool = multiprocessing.pool.ThreadPool(N)
    for x in pool.imap(_single_compile, objects): pass
    return objects
import distutils.ccompiler
distutils.ccompiler.CCompiler.compile=parallelCCompile
```

pipriles, 2018-11-02:
If i try to reproduce it with a dummy function:

```python
In [10]: def hey(objs):
    ...:     list(mp.ThreadPool(8).imap(fn, objs))
    ...:     return objs
```

And then calling it and doing `Ctrl + C` to see where it hangs...

```python
----> 1 test([1, 2, 3])

<ipython-input-24-bfcf20059a93> in test(objs)
      1 def test(objs):
----> 2     list(mp.ThreadPool(8).imap(fn, objs))
      3     return objs

/usr/lib/python3.7/multiprocessing/pool.py in next(self, timeout)
    761                 if self._index == self._length:
    762                     raise StopIteration from None
--> 763                 self._cond.wait(timeout)
    764                 try:
    765                     item = self._items.popleft()
```

youssefzaky, 2018-11-08:
I've had the same issue on MacOS mojave and Ubuntu 18. It works after removing that monkey patch

erwincoumans, 2018-11-08:
@youssefzaky could you try attached zipfile? It applies a workaround suggested by @pipriles 

[setup.zip](https://github.com/bulletphysics/bullet3/files/2562856/setup.zip)
~~~~~~~~~~~~~~~~~~
git clone https://github.com/bulletphysics/bullet3
cd bullet3
unzip ~/Downloads/setup.zip
python setup build
python setup install --user
~~~~~~~~~~~~~~~~~~

erwincoumans, 2018-11-08:
Great, thanks for testing. I'll fix it.

erwincoumans, 2018-11-08:
Will be fixed in this PR: https://github.com/bulletphysics/bullet3/pull/1985