bulletphysics/bullet3 #1958: pybullet robot simulation go crazy!!!!!

Frank-Li-Code, 2018-10-27:
I use official examples: [https://github.com/bulletphysics/bullet3/tree/master/examples/pybullet/tensorflow](url)
Every time when robot velocity is little fast or it collision on it self, then it go crazy like this:
[image]
When have this issues, there just no way to training agent to do anything at all!!! : (

erwincoumans, 2018-10-27:
The demo settings weren't tweaked to deal with additional energy of mouse picking etc. It would be good to make it a bit more robust though, here is a PR to improve it:
https://github.com/bulletphysics/bullet3/pull/1959
Thanks for the report!

closed 2018-10-27 by erwincoumans

erwincoumans, 2018-10-27:
Another issue is that the resetDebugVisualizerCamera call will interfere with mouse picking, adding even more energy into the system. If you want to pick, best to disable the call to resetDebugVisualizerCamera in that demo. I'll see if we can either disable picking in that demo, or remove the follow cam.