bulletphysics/bullet3 #1963: Do the pyBullet AI training baselines need updating?

mattngc, 2018-10-29:
I'm having trouble running some of the AI deep Qlearning examples for pybullet. It appears some of these demo files are a little out of date. Take for example train_pybullet_cartpole.py:
https://github.com/bulletphysics/bullet3/blob/master/examples/pybullet/gym/pybullet_envs/baselines/train_pybullet_cartpole.py

When I first ran it I ran into an error in the deepq.learn() function, it was missing a model positional argument. I quickly fixed that by changing the example slightly to

```python
    act = deepq.learn(
        env,
        model,
        lr=1e-3,
        max_timesteps=100000,
        buffer_size=50000,
        exploration_fraction=0.1,
        exploration_final_eps=0.02,
        print_freq=10,
        callback=callback
    )
```

But then I ran into the next error that might require deeper knowledge of the baselines package to fix. 
```
$> python train_pybullet_cartpole.py                                         
pybullet build time: Oct 28 2018 19:13:43
current_dir=...
Logging to /tmp/openai-2018-10-29-18-40-51-146989
WARN: gym.spaces.Box autodetected dtype as <class 'numpy.float32'>. Please provide explicit dtype.
2018-10-29 18:40:51.289705: I tensorflow/core/platform/cpu_feature_guard.cc:141] Your CPU supports instructions that this TensorFlow binary was not compiled to use: FMA
---------------------------------------------------------------------------
TypeError                                 Traceback (most recent call last)
~/train_pybullet_cartpole.py in <module>
     40 
     41 if __name__ == '__main__':
---> 42     main()

~/train_pybullet_cartpole.py in main()
     33         exploration_final_eps=0.02,
     34         print_freq=10,
---> 35         callback=callback
     36     )
     37     print("Saving model to cartpole_model.pkl")

~/baselines/baselines/deepq/deepq.py in learn(env, network, seed, lr, total_timesteps, buffer_size, exploration_fraction, exploration_final_eps, train_freq, batch_size, print_freq, checkpoint_freq, checkpoint_path, learning_starts, gamma, target_network_update_freq, prioritized_replay, prioritized_replay_alpha, prioritized_replay_beta0, prioritized_replay_beta_iters, prioritized_replay_eps, param_noise, callback, load_path, **network_kwargs)
    207         gamma=gamma,
    208         grad_norm_clipping=10,
--> 209         param_noise=param_noise
    210     )
    211 

~/baselines/baselines/deepq/build_graph.py in build_train(make_obs_ph, q_func, num_actions, optimizer, grad_norm_clipping, gamma, double_q, scope, reuse, param_noise, param_noise_filter_func)
    374             param_noise_filter_func=param_noise_filter_func)
    375     else:
--> 376         act_f = build_act(make_obs_ph, q_func, num_actions, scope=scope, reuse=reuse)
    377 
    378     with tf.variable_scope(scope, reuse=reuse):

~/baselines/baselines/deepq/build_graph.py in build_act(make_obs_ph, q_func, num_actions, scope, reuse)
    181         eps = tf.get_variable("eps", (), initializer=tf.constant_initializer(0))
    182 
--> 183         q_values = q_func(observations_ph.get(), num_actions, scope="q_func")
    184         deterministic_actions = tf.argmax(q_values, axis=1)
    185 

~/baselines/baselines/deepq/models.py in q_func_builder(input_placeholder, num_actions, scope, reuse)
     99     def q_func_builder(input_placeholder, num_actions, scope, reuse=False):
    100         with tf.variable_scope(scope, reuse=reuse):
--> 101             latent = network(input_placeholder)
    102             if isinstance(latent, tuple):
    103                 if latent[1] is not None:

~/baselines/baselines/deepq/models.py in <lambda>(*args, **kwargs)
     28         q_function for DQN algorithm.
     29     """
---> 30     return lambda *args, **kwargs: _mlp(hiddens, layer_norm=layer_norm, *args, **kwargs)
     31 
     32 

TypeError: _mlp() missing 2 required positional arguments: 'num_actions' and 'scope'
```

Any help with getting these demos up-to-date and working would be appreciated.

I've installed Bullet and pyBullet through pip and anaconda. I'm using:
- Python 3.6.6
- pybullet 2.3.4
- tensorflow 1.11.0
- gym 0.10.8
- baselines 0.1.5

Thanks for your help in advance :)

erwincoumans, 2018-10-29:
Demos are up-to-date with baselines 0.1.3. 

OpenAI baselines and TensorFlow often break the API, could it be the issue?

erwincoumans, 2018-10-29:
I just checked the cartpole example and made some updates to make it more similar to the Gym classical control cartpole. https://github.com/bulletphysics/bullet3/pull/1962
Will be in pip pybullet 2.3.5 and later or just use github version.
It works just fine with latest baselines 0.1.5, using python 3.7 on Mac OSX.
Note that baselines doesn't work with Python 2.7!
Can you check it?

closed 2018-10-29 by erwincoumans

mattngc, 2018-10-30:
thanks for the fast reply.

hmm... I've upgraded to pybullet 2.3.5 as per your instruction, but unfortunately I'm still getting the same error when I run this script.
https://github.com/bulletphysics/bullet3/blob/master/examples/pybullet/gym/pybullet_envs/baselines/train_pybullet_cartpole.py

```
matt@matt-amdPC:~/$ python train_pybullet_cartpole.py 
pybullet build time: Oct 30 2018 16:23:38
current_dir=...
Logging to /tmp/openai-2018-10-30-16-38-55-224342
2018-10-30 16:38:55.239391: I tensorflow/core/platform/cpu_feature_guard.cc:141] Your CPU supports instructions that this TensorFlow binary was not compiled to use: FMA
Traceback (most recent call last):
  File "train_pybullet_cartpole.py", line 39, in <module>
    main()
  File "train_pybullet_cartpole.py", line 32, in main
    callback=callback
  File "/home/matt/baselines/baselines/deepq/deepq.py", line 209, in learn
    param_noise=param_noise
  File "/home/matt/baselines/baselines/deepq/build_graph.py", line 376, in build_train
    act_f = build_act(make_obs_ph, q_func, num_actions, scope=scope, reuse=reuse)
  File "/home/matt/baselines/baselines/deepq/build_graph.py", line 183, in build_act
    q_values = q_func(observations_ph.get(), num_actions, scope="q_func")
  File "/home/matt/baselines/baselines/deepq/models.py", line 101, in q_func_builder
    latent = network(input_placeholder)
  File "/home/matt/baselines/baselines/deepq/models.py", line 30, in <lambda>
    return lambda *args, **kwargs: _mlp(hiddens, layer_norm=layer_norm, *args, **kwargs)
TypeError: _mlp() missing 2 required positional arguments: 'num_actions' and 'scope'
```

Here are all the versions I'm using.
python                    3.6.6                
pybullet                  2.3.5                     <pip>
baselines                 0.1.5                     <pip>
gym                       0.10.8                    <pip>

Maybe I need to update to the latest pybullet github version to get the changes you made in #1962? Also, I'm running this file from a local directory, not inside the module directory. Would this matter? Cause I notice you are doing some path manipulations at the top of the file.

erwincoumans, 2018-10-30:
Yes, update the github version. It seems more like a baselines issue. Do other environments work well?

mattngc, 2018-10-31:
matt@amdPC:~$ python --version
Python 3.6.6 :: Anaconda, Inc.

Alright, I'll see if I can update to the github version. I'm on Ubuntu 18.04.

I'll also try diving into baselines to see if I can understand the problem. I'm fairly experienced with python, but just starting to learn the AI packages.

mattngc, 2018-10-31:
Ok, you are definitely right. I purged my environment of baselines and gym. Re-installed both and now I can run your demos. I think I must have at one point have installed baselines from master on github, which has changes that produce the bug I was seeing. Anyway, resolved now. Thanks for you help and sorry for wasting your time.