bulletphysics/bullet3 #1966: Is this line from PyBullet okay?

dustContributor, 2018-10-31:
So [LWJGL](https://github.com/LWJGL/lwjgl3) project made Java bindings for Bullet's C API, reading pybullet.c sources to check how it's used I came across this line:

https://github.com/bulletphysics/bullet3/blob/master/examples/pybullet/pybullet.c#L6467

Shouldn't it be b3CreateCollisionShapeSetChildTransform? Instead of the visual one.

erwincoumans, 2018-10-31:
Correct. Luckily at the moment, this is the implementation:
~~~~~~~~~~

B3_SHARED_API void b3CreateVisualShapeSetChildTransform(b3SharedMemoryCommandHandle commandHandle, int shapeIndex, const double childPosition[/*3*/], const double childOrientation[/*4*/])
{
	b3CreateCollisionShapeSetChildTransform(commandHandle, shapeIndex, childPosition, childOrientation);
}
~~~~~~~~~~~~

dustContributor, 2018-10-31:
I'll close this since you fixed it in 47b8b98a64ca1159d19555ff985ca7952b24be60

Thanks! And thanks a lot for your work on Bullet since we're at it! :smile:

closed 2018-10-31 by dustContributor