bulletphysics/bullet3 #1970: Wireframe drawing for btConvexHullShape is incorrect.

stolk, 2018-10-31:
Start up "convex stack" Benchmark Demo.

Press W.

The "Taru" shape shows up incorrectly when in wireframe, but correctly when it draws the full faces.

[image]

[image]

If I had to guess: it seems to be drawing indexed-vertex data as if it was not indexed?

erwincoumans, 2018-10-31:
This is not a bug but expected. The convex doesn't have indices, so there is nothing to draw. You can force to generate faces for prettier wireframe if you like.

~~~~~~~~~
convexHull->initializePolyhedralFeatures();
~~~~~~~~~

closed 2018-10-31 by erwincoumans

erwincoumans, 2018-11-01:
Will be fixed soon anyway: https://github.com/bulletphysics/bullet3/commit/9e58a217324d555bb11eb5194825906f977c82e6