bulletphysics/bullet3 #1971: [C API] Failing to read translation and orientation from a collision mesh

dustContributor, 2018-10-31:
Hi! I am using [LWJGL](https://github.com/LWJGL/lwjgl3)'s Java bindings for the C API and I am unable to fetch the translation and orientation data from a collision shape. b3GetStatusType returns invalid after submitting the command.

```java
public static final void main ( final String[] args ) {
	// World transform that we want to refresh with Bullet's 
	var world = new Mat4f();
	var phy = b3ConnectPhysicsDirect();
	if ( phy == MemoryUtil.NULL ) {
		throw new IllegalStateException();
	}
	{
		long cmd = b3InitPhysicsParamCommand( phy );
		b3PhysicsParamSetGravity( cmd, 0, 0, -9.8 );
		b3PhysicsParamSetTimeStep( cmd, 1.0 / 60.0 );
		var status = b3SubmitClientCommandAndWaitStatus( phy, cmd );
		var type = b3GetStatusType( status );
		if ( type != CMD_CLIENT_COMMAND_COMPLETED ) {
			throw new IllegalStateException();
		}
	}
	int uid;
	{
		var cmd = b3CreateCollisionShapeCommandInit( phy );
		var index = b3CreateCollisionShapeAddCapsule( cmd, 1, 1 );
		try ( var mem = MemoryStack.stackPush() ) {
			var pos = new Vector3f( 1, 0, 0 );
			world.rotateLocalX( 45 );
			world.translate( pos );
			var quat = new Quaternionf().setFromNormalized( world );
			b3CreateCollisionShapeSetChildTransform( cmd, index,
					mem.mallocDouble( 3 )
							.put( 0, pos.x )
							.put( 1, pos.y )
							.put( 2, pos.z )
							.rewind(),
					mem.mallocDouble( 4 )
							.put( 0, quat.x )
							.put( 1, quat.y )
							.put( 2, quat.z )
							.put( 3, quat.w )
							.rewind() );
		}
		var handle = b3SubmitClientCommandAndWaitStatus( phy, cmd );
		var type = b3GetStatusType( handle );
		if ( type != CMD_CREATE_COLLISION_SHAPE_COMPLETED ) {
			throw new IllegalStateException();
		}
		uid = b3GetStatusCollisionShapeUniqueId( handle );
	}
	{
		var cmd = b3RequestActualStateCommandInit( phy, uid );
		var status = b3SubmitClientCommandAndWaitStatus( phy, cmd );
		var type = b3GetStatusType( status );
		// Here is where it fails.
		if ( type != CMD_ACTUAL_STATE_UPDATE_COMPLETED ) {
			throw new IllegalStateException();
		}
		try ( var mem = MemoryStack.stackPush() ) {
			var ptr = mem.mallocPointer( 1 );
			b3GetStatusActualState( status, null, null, null, null, ptr, null, null );
			// Size is 3 doubles for position + 4 for orientation.
			var data = ptr.getDoubleBuffer( 0, 7 );
			// If we ignore the status, these make the VM crash because ptr isn't valid.
			var pos = new Vec3f( 
					(float) data.get(), 
					(float) data.get(), 
					(float) data.get() );
			var or = new Quaternionf( 
					(float) data.get(), 
					(float) data.get(), 
					(float) data.get(), 
					(float) data.get() );
			// Read back Bullet's data into our world transform object.
			world.set( or ).setTranslation( pos );
		}
	}
	System.out.println( "Finished!" );
	System.out.println( world );
}
```

The Bullet calls are straightforward and a fairly direct translation from the C API, I could try to do a C program that exhibits the same issue if it helps but I don't have experience with building C programs so it might take a while.

I've reported the issue with LWJGL's maintainers but they claim the bindings are okay.

erwincoumans, 2018-10-31:
Please create a PyBullet python script as reproduction.

erwincoumans, 2018-10-31:
b3RequestActualStateCommandInit requires an object unique id, not a shape unique id. Shapes don't have transforms.

closed 2018-10-31 by erwincoumans

dustContributor, 2018-11-01:
Ah right! I completely forgot that you had to create a collision shape, and _then_ a rigid body out of the collision shape(s).

So I suppose I should be looking at b3CreateMultiBody* and friends. I'll try that out, thanks!

erwincoumans, 2018-11-01:
I would look at the PyBullet examples and pybullet.c.
An alternative is wrapping the b3RobotSimulatorClientAPI_NoDirect.h API, this C++ API already has most of the stuff implemented, on top of the C API.

dustContributor, 2018-11-01:
> By default, we don't programmatically create things in PyBullet/robotics, but we load from URDF. I would focus on that too, because creating programmatically is very hard and error-prone

Ah right, I never mentioned, this is a game oriented project. I'm going to use all the pre-defined shapes (say, capsules are popular for modelling player entities, spheres and boxes are useful low poly approximations for a lot of things), plus whatever convex mesh I need (for which I needed the new API to make collision shapes out of vertex lists). The format for storing these things will be most probably be custom and/or some widely used format for 3D data (fbx, collada, maybe obj who knows).

I haven't got that far yet but I'm also planning on using Bullet for trigger zones and hit boxes, which can be used for raycasting (or enter/leave events) but don't necessarily "collide" with normal stuff, so I am probably also going to make use of the collision masks you expose through a plugin for the C API.

I wont probably touch the "visual" API at all given I have my own renderer.

I know the API is robotics focused but I'm hoping it will be enough for my use case, mine is a hobby project so I don't need AAA game performance/flexibility out of it, just the usual things for a first person game. Of course it'd be nice to have a "full" Bullet C API and use that instead but given Bullet is a mature project and a direct C API doesn't exists, I gather it isn't a priority or there isn't enough interest or time to do it so I don't mind.

> An alternative is wrapping the b3RobotSimulatorClientAPI_NoDirect.h API, this C++ API already has most of the stuff implemented, on top of the C API.

Thanks for the suggestion, I'm exclusively using the C API directly here because LWJGL's author has been kind enough to do bindings for it. He doesn't does C++ bindings, and I'm not really into that sort of project to do it on my own.