bulletphysics/bullet3 #1973: question about DIRECT model

lyz22233, 2018-11-01:
I set p.connect(p.DIRECT), and I use changeConstraint to change Obj's position, then I use  getBasePositionAndOrientation to get the obj's position and print it , but it has no change.

when I set p.connect(p.GUI), other code remains unchanged，then it's position can change.

erwincoumans, 2018-11-01:
changeConstraint doesn't change the position. You need to step the simulation (stepSimulation). Did you do that?

lyz22233, 2018-11-01:
Problem solved，thank you!

lyz22233, 2018-11-01:
> changeConstraint doesn't change the position. You need to step the simulation (stepSimulation). Did you do that?

Problem solved，thank you!

erwincoumans, 2018-11-01:
realTimeSimulation gets update from a separate thread, hance not DIRECT.

closed 2018-11-01 by erwincoumans